epics-modules/motor #8: Raw positions are double in driver but integer in record

alex-soderqvist, 2015-11-24:
Raw positions, motorPosition_ and motorEncoderPosition_, in the asynPortDriver are [doubles](https://github.com/epics-modules/motor/blob/master/motorApp/MotorSrc/asynMotorAxis.cpp#L274).

In the record these parameters are [long integers](https://github.com/epics-modules/motor/blob/master/motorApp/MotorSrc/motorRecord.dbd#L573).

The conversion is done in the [device support](https://github.com/epics-modules/motor/blob/master/motorApp/MotorSrc/devMotorAsyn.c#L423).

It should be clear that this is an integer so that you handle it in the driver, if not scaled to an integer this have major implications!
The record reports a rounded value as "raw" and for example retries does not work properly.

I am using TPMAC model 3 driver from DLS. On the Delta Tau PMAC motion controller family the positions are reported as as floating point values, but rounded to nearest 1/32 fraction. But no scaling and conversion is done in the driver so that this is represented as an integer, since you seem to be setting a double precision value.

I suggest that you change the parameters in asynPortDriver to integers, so that it is clear that it has to be converted. In this case a multiplication by 32 and a type conversion.

MarkRivers, 2015-11-24:
I disagree with this suggestion.  Model 3 drivers are not limited to being used with the motor record, they can be used with generic asyn device support or potentially another framework.  The limitation of the RVAL field in the motor record being an epicsInt32 is potentially a serious restriction that should be changed sometime in the future.  It potentially limits the travel range of a device with high resolution.  The motor record passes all target positions for move absolute, move relative, set position, etc. as doubles anyway.

alex-soderqvist, 2015-11-24:
I agree, I shouldn't have been that fast to post a suggestion. I of course agree that it is very bad to limit RVAL to an epicsInt32. But the issue is till valid. Thanks for pointing out the real reason.

Although, this leaves me with no good solution since modifying the driver would be adapting to a problem of the motor record.

MarkRivers, 2015-11-24:
What is the problem you are having?  I use a number of motor controllers that use floating point positions (Newport XPS, Delta Tau GeoBrick) and have not run into problems.

alex-soderqvist, 2015-11-24:
I noticed this with soft limits. The PMAC family reports this soft limits and hard limits in the same way. So when moving to the soft limit from the motor record, lets say limit in RAW values is 1000 and you order a move to RVAL 1000, it would not always reach this position, instead it reached 999.5 but RMP shows 1000. And HIGH limit bit in MSTA would not show that it reached the maximum limit, and it would be a mystery to the user why it didn't show that since RMP and RBV is 1000.

I mean that this rounding in the device support has implications, it is bad that it does show the true raw value. And if you take into account follow up problems, limits and retries etc. It may potentially be problems when you try to build logic around it.

MarkRivers, 2015-11-24:
Can you just scale up the positions so they always are an integer?  That is what I do with the XPS.  There is a user-settable scale factor for each axis.  This scales the XPS float value before it is converted to an integer.  If it is set to the encoder resolution then the result will always be an integer.

alex-soderqvist, 2015-11-25:
Yes, I can and wait until someone fixes the types in the motor record.

Note that the documentation is wrong, it states that the RMP and ERES indeed are doubles.

http://www.aps.anl.gov/bcda/synApps/motor/R6-9/motorRecord.html

MarkRivers, 2015-11-25:
This is from motorRecord.h

        epicsFloat64    eres;   /* Encoder Step Size (EGU) */
        epicsInt32      rmp;    /* Raw Motor Position */

So ERES is double, but RMP is int32.

alex-soderqvist, 2015-11-25:
I am sorry I meant Raw Encoder position, REP. 
Thanks for helping me solve the problem.

Please close the issue whenever all fields in the motor record that should doubles have been converted to that.

MarkRivers, 2015-11-25:
I'm not sure if those fields will ever actually be changed, since it might break a lot of things.

closed 2018-05-02 by kmpeters