epics-modules/motor #10: motor module documentation

prjemian, 2015-11-25:
Try to host all module documentation either in the <module>/documentation or in the module's GitHub wiki.  Any person with write permission can update/modify the documentation as needed.

prjemian, 2015-11-25:
This issue started in emails.  Here is history:

@MarkRivers wrote:
For many synApps modules that "top-level" Web page is not called index.html rather it is called XXX.html, where XXX is the module name.  For example, mca.html, ip330.html, etc.  I think this is nicer than index.html, because if one copies the motor/documentation tree into some Web server, you don't have to blow away some existing index.html file.  I ran into this exact problem (not with motor) yesterday.

Is there a convenient way to display HTML files on github without having to manually prefix them with htmlpreview.github.io?  How did you generate that link you sent us? 

I wrote:
Let's take your suggestion and change the name of <motor>/documentation/index.html to <motor>/documentation/motor.html   (60a2298)

prjemian, 2015-11-25:
We are using the most popular method that I can find easily. See this discussion:

http://stackoverflow.com/questions/8446218/how-to-see-an-html-page-on-github-as-a-normal-rendered-html-page-to-see-preview

Another SO page (http://stackoverflow.com/questions/6551446/can-i-run-html-files-directly-from-github-instead-of-just-viewing-their-source?lq=1) offers the suggestion to use this (shorter, clearer) pattern:
```
https://rawgit.com/[user]/[repository]/[branch]/[filename.ext]
```

In the case of the motor.html page:

https://rawgit.com/epics-modules/motor/master/documentation/motor.html

Comments?

MarkRivers, 2015-11-25:
I don't think that users should have to use rawgit and cut and paste to access documentation.  It should be just as easy as it is now.  I also don't want to have to change the links in all documentation, as it appears that you did for the motorRecord.html links in motor.html?  Until github will directly render HTML I think we should use github as the source repository for documentation, but install it on a Web server where no cut and paste is required, and where existing relative links will work. I have lots of documentation that assumes that related documentation is present in the same directory as the html file that loads the documentation.  For example, areaDetectorDoc.html assumes that prosilicaDoc.html is present in the same directory on the Web server, even though it is not in the same source code directory.

kmpeters, 2018-08-21:
Github pages allows us to have html documentation that works online and locally:

  https://epics-modules.github.io/motor/

I had to rename the "documentation" directory to "docs" (fd4641f2671ffa6d28ab031485547c42bb134637) so that github pages could be enabled.  I also had to rename motor.html to index.html (3b75e3096f0fc1a090769c03d7273b3cf2c4f42b)

closed 2018-08-21 by kmpeters