epics-modules/motor #11: missing tags

prjemian, 2015-11-25:
According to the motor main page (http://htmlpreview.github.io/?https://github.com/epics-modules/motor/blob/master/documentation/motor.html), the oldest tag shown for this repository is R6-5-2 (Oct 7, 2010).

prjemian, 2015-11-25:
The complete tag list for motor, up to the GIT tag:
https://subversion.xray.aps.anl.gov/trac/synApps/browser/motor/tags

The script used for the conversion:
https://github.com/BCDA-APS/conversions/blob/master/svn_to_github.sh

prjemian, 2015-11-25:
http://stackoverflow.com/questions/23555582/set-a-specific-date-for-a-git-tag

get the hash numbers and dates by logging output
```
./svn_to_github.sh motor | tee log.txt
grep /tags/ log.txt
```

Taking that output and editing a text file, get commands such as:
```
GIT_COMMITTER_DATE="2000-09-06 14:18:22" git tag -a Version4-1 -m Version4-1 c714129b1d5ec765a7939bff96847ae22b437be0
```

prjemian, 2015-11-25:
This gets the short hash and date in ISO8601 format for a given hash
```
git show -s --pretty=format:"%h %ci" c714129b1d5ec765a7939bff96847ae22b437be0
```

prjemian, 2015-11-25:
Apparently, the tags created by cvs2svn have not been converted to GIT

```
[linux] $ git show -s 1335b1617812f058340ad6be69aced586ecffc04
commit 1335b1617812f058340ad6be69aced586ecffc04
Author: cvs2svn <rsluiter@users.noreply.github.com>
Date:   Thu Jul 18 19:14:14 2002 +0000

    This commit was manufactured by cvs2svn to create tag 'R3_13_base'.
```

kmpeters, 2015-11-25:
I see that the email in this commit shows rsluiter as being cvs2svn, but in the users.txt file I believe was used for the svn-to-git conversion, cvs2svn is prjemian.  Could this mismatch be the source of the problem?

prjemian, 2015-11-25:
Perhaps but I doubt it.  The users.txt file changed since motor was moved.

The hashes listed above (c714129b1d5ec765a7939bff96847ae22b437be0, 1335b1617812f058340ad6be69aced586ecffc04) are not found in the current repository.  That's the problem.  

Walking through the conversion script line-by-line, the tags by cvs2svn are not identified at this line (while later tags, done in svn, *are* identified):

```
git svn clone $SYNAPPSSVNURL/$1 --authors-file=users.txt --no-metadata -s $1
```

prjemian, 2015-11-26:
If commits with the associated hash ids were present, then it would be possible to make these tags retroactively.  This bash script would do it:

```bash
#!/bin/sh

GIT_COMMITTER_DATE="2000-04-19 15:40:05"  git tag -a   Version4-1	    -m   Version4-1	      c714129b1d5ec765a7939bff96847ae22b437be0
GIT_COMMITTER_DATE="2000-09-06 14:18:22"  git tag -a   R4-2		    -m   R4-2		      36d61447d162de2aafa70c717abc69b196a39e5f
GIT_COMMITTER_DATE="2001-02-14 17:09:41"  git tag -a   R4-3		    -m   R4-3		      902456c3581b8aeb3c97497123935df25cc57f7a
GIT_COMMITTER_DATE="2001-08-22 19:32:39"  git tag -a   R4-4		    -m   R4-4		      11f08b6db36b9df8735bf6b8bb2efb0fe3fd6f12
GIT_COMMITTER_DATE="2002-02-18 17:18:40"  git tag -a   R4-3-1		    -m   R4-3-1 	      4a6c6debc537574026f83a274b100f1f42466d73
GIT_COMMITTER_DATE="2002-02-18 19:54:42"  git tag -a   R4-4-1		    -m   R4-4-1 	      1fb5e466675420e145dab1e2317c45ad6e3e6d96
GIT_COMMITTER_DATE="2002-02-20 20:47:57"  git tag -a   R4-3-2		    -m   R4-3-2 	      eeb89855c15252b15970a40c51125795b33805bd
GIT_COMMITTER_DATE="2002-07-18 19:14:14"  git tag -a   R3_13_base	    -m   R3_13_base	      1335b1617812f058340ad6be69aced586ecffc04
GIT_COMMITTER_DATE="2002-07-18 19:14:15"  git tag -a   R4-5		    -m   R4-5		      fe3dd1752e6cd46583e684d0b935d92251a7c4d8
GIT_COMMITTER_DATE="2002-10-01 18:35:06"  git tag -a   R4-4-2		    -m   R4-4-2 	      8a26a38881022c22aa1a69b9b5ee76c354ddcca2
GIT_COMMITTER_DATE="2002-10-29 19:04:36"  git tag -a   R4-5-1		    -m   R4-5-1 	      fe38f9606467d5522097286898c3fd0278877234
GIT_COMMITTER_DATE="2003-02-26 16:54:45"  git tag -a   R4-6		    -m   R4-6		      118db3972ba5c14c188d5940236b7fbf3c944d81
GIT_COMMITTER_DATE="2003-03-03 17:04:41"  git tag -a   R4-6-1		    -m   R4-6-1 	      a3a2eb9139e55029d383da945c150960314ac5da
GIT_COMMITTER_DATE="2003-04-24 19:39:03"  git tag -a   R4-7		    -m   R4-7		      d7c4023ab126c110d136d79d9de5313620437958
GIT_COMMITTER_DATE="2003-06-02 21:55:51"  git tag -a   R4-5-2		    -m   R4-5-2 	      1cd58b211db3c0823d9c89da79dd1f4e9c77e150
GIT_COMMITTER_DATE="2003-06-03 14:49:24"  git tag -a   R4-6-2		    -m   R4-6-2 	      b4133ec858675780d54155ec3f1339c001401852
GIT_COMMITTER_DATE="2003-06-03 18:07:21"  git tag -a   R4-7-1		    -m   R4-7-1 	      879441fd62cafeb37e1e388e99423917f9521240
GIT_COMMITTER_DATE="2003-06-16 15:42:08"  git tag -a   R5-1		    -m   R5-1		      78ac3c6a484d39ae6f0fcdb308e417d86dfb7770
GIT_COMMITTER_DATE="2003-10-28 19:15:17"  git tag -a   R4-6-3		    -m   R4-6-3 	      79051ed9f045c7a50a06025f491206d566745245
GIT_COMMITTER_DATE="2003-10-29 20:07:26"  git tag -a   R4-7-2		    -m   R4-7-2 	      01dfe21188991bd062eac91ba0fcf7e36f1e0a57
GIT_COMMITTER_DATE="2003-10-30 21:40:42"  git tag -a   R4-5-3		    -m   R4-5-3 	      f9a2328468fcff9658be9afa2707207f93ce7966
GIT_COMMITTER_DATE="2003-10-31 16:51:44"  git tag -a   R4-6-4		    -m   R4-6-4 	      f3e0fce3b38fb41043dd418a122b9a7997ede1e6
GIT_COMMITTER_DATE="2003-11-03 16:24:12"  git tag -a   R4-7-3		    -m   R4-7-3 	      3c4b305feae9a2b68162a4ef794c0dbd0178cd8f
GIT_COMMITTER_DATE="2003-11-10 15:29:31"  git tag -a   R5-2		    -m   R5-2		      781a7a8659dd55003f775608939bd933efb1d7c6
GIT_COMMITTER_DATE="2003-12-11 17:40:07"  git tag -a   R4-7-4		    -m   R4-7-4 	      610e14c1e29b7d1834ba435e3cd2009a1693f15f
GIT_COMMITTER_DATE="2003-12-17 20:07:51"  git tag -a   R4-8		    -m   R4-8		      2f4ba5e0d610d7888850284958cce98c989fdeec
GIT_COMMITTER_DATE="2003-12-17 20:07:52"  git tag -a   synApps_4_6	    -m   synApps_4_6	      e0132e241ff3b4a3e590ae24554433119400de66
GIT_COMMITTER_DATE="2004-01-28 16:35:02"  git tag -a   R4-9		    -m   R4-9		      e21faf1015f5a3d82f942f985bef2fd55d651dc5
GIT_COMMITTER_DATE="2004-02-27 20:02:54"  git tag -a   R4-5-4		    -m   R4-5-4 	      363a61889e463cf8258074e64f5e639f869a1a5b
GIT_COMMITTER_DATE="2004-03-16 20:59:12"  git tag -a   R5-3		    -m   R5-3		      97404520fe3b25a90d29588c334878ac7f4ec604
GIT_COMMITTER_DATE="2004-03-23 20:13:02"  git tag -a   R4-9-1		    -m   R4-9-1 	      a271dd9eb20cbf4761562ea693ac4aec655974e5
GIT_COMMITTER_DATE="2004-05-03 16:16:41"  git tag -a   R4-9-2		    -m   R4-9-2 	      14bb8ea620255bce4527e1f34c49e1c569d188d0
GIT_COMMITTER_DATE="2004-05-24 18:31:23"  git tag -a   R4-8-1		    -m   R4-8-1 	      597495beedc1d75aa34bcc80c90770b539569f95
GIT_COMMITTER_DATE="2004-05-25 16:49:43"  git tag -a   R4-9-3		    -m   R4-9-3 	      33dcd0119d3a59549d0f5f427bbef3afe5d383f8
GIT_COMMITTER_DATE="2004-09-21 15:33:18"  git tag -a   Diamond_branch_base  -m   Diamond_branch_base  facac749bd057a8e4278946bcbda34e59ff1f7a4
GIT_COMMITTER_DATE="2004-09-21 15:33:19"  git tag -a   R4-9-4		    -m   R4-9-4 	      450b73559fb051c195e89d68fc3608fd01f6ce6d
GIT_COMMITTER_DATE="2004-09-21 18:22:49"  git tag -a   R5-4		    -m   R5-4		      d29946b6f208a9aea054def3846df25ddd45e649
GIT_COMMITTER_DATE="2004-12-21 20:18:32"  git tag -a   R5-5		    -m   R5-5		      53fd6581b4942d6bfbaf319668b9c411a902e243
GIT_COMMITTER_DATE="2005-03-30 22:24:13"  git tag -a   R5-6		    -m   R5-6		      e0451b2f3b44f3744c78ffbe1d6fc9f51ca33956
GIT_COMMITTER_DATE="2005-03-30 22:24:14"  git tag -a   synApps_5_1	    -m   synApps_5_1	      b2743b869ecc1ecb9e48fdfe4dbe435512a2b835
GIT_COMMITTER_DATE="2005-04-22 18:41:24"  git tag -a   R4-9-5		    -m   R4-9-5 	      89fccd6e727f3b1724aaebd1f1176fe825d5c54f
GIT_COMMITTER_DATE="2005-10-26 19:17:51"  git tag -a   R4-9-6		    -m   R4-9-6 	      4b24532710046f747ec0dd5a503c988c6fb90d19
[... cut ...]
GIT_COMMITTER_DATE="2009-06-22 18:53:05"  git tag -a   R6-4-4		    -m   R6-4-4 	      05a436c686655fdb68e3deee8185ccbb260a31e9
GIT_COMMITTER_DATE="2010-04-21 21:05:04"  git tag -a   R6-5		    -m   R6-5		      5d3ff712fe0e288dd08d5d1603c3700c67228fec
GIT_COMMITTER_DATE="2010-04-21 21:11:41"  git tag -a   synApps_5_5	    -m   synApps_5_5	      4a38e41b2eb1cc6c8e7ff1601689c2e4651ec17a
GIT_COMMITTER_DATE="2010-06-09 20:32:58"  git tag -a   R6-5-1		    -m   R6-5-1 	      2f6198265a6f1958a39bd2d096cf4cd9e13d0207
```

prjemian, 2015-11-26:
Summary:
Cannot apply known tags from SVN retroactively to this repository since commits with the corresponding hash ids are missing.  Changes to the source code have been spot-checked and found to be intact.  The only way to gain these older tags is by rebuilding this repository from subversion.  Removal and rebuilding the repository would be disruptive to the subsequent GitHub change history, the issue history, and to those who have clone or checked out the module from GitHub.

Polled the main module contributors and they all agree to accept the repository as it exists now, rather than rebuild it to include the older tags.

closed 2015-11-26 by prjemian