epics-modules/motor #9: <motor>/README looks out of place

prjemian, 2015-11-25:
This HTML page contains release notes and known problems for current development and past releases.  Shouldn't this go into the documentation subdirectory?

kmpeters, 2015-11-25:
My understanding is that the <motor>/README contains details about changes to the motor module that only developers would care about.  The release notes HTML page contains a higher-level view of the same changes and is intended for users of the module.

prjemian, 2017-02-07:
We should describe this clearly on the README.md page that appears on the GH repository landing page.

kmpeters, 2018-08-21:
I added a note to the README in commit 497bdac0c9e6e5c9e7873157cdeb262ce15b3358 letting people know that the file won't be updated in the future.  This feels like a better place to document it than the motor module landing page; I'd rather not draw attention to a file that will not be maintained in the future.

closed 2018-08-21 by kmpeters