epics-modules/motor #12: Softlimits when jogging and DIR=1 does not work

tboegi, 2015-12-03:
The softlimits don't work, when jogging with DIR=1

A Solution is propoosed here:
https://github.com/epics-modules/motor/pull/1

rsluiter, 2016-02-22:
Hello Torsten,

I don't know if you received an email notification of my pull request.
I have an alternative solution here; [18](https://github.com/epics-modules/motor/pull/18)
Please read the comment under "Conversation".

Ron

tboegi, 2016-02-23:
Yes, I got all the emails, apologies  for not answering earlier.
Thanks for picking this up, improving and testing.

I will test it on our Model 3 test system, which should be completed the next days or so.

tboegi, 2016-06-07:
The real fix is here:
https://github.com/epics-modules/motor/pull/18
https://github.com/epics-modules/motor/commit/4efe15e76cefdc060e14dbc334ad8360b2858763

closed 2016-06-07 by tboegi