epics-modules/motor #25: Motor record is issuing stops if RA_PROBLEM bit is set

mp49, 2016-04-05:
The following change to the motor record broke the XPS home functions:

Set "stop" field true if driver returns RA_PROBLEM true. (Motor record
stops motion when controller signals error but does not stop motion; e.g.,
maximum velocity exceeded.)

The relevant bit of code seems to be:

 /* Get motor-now-moving indicator. */
    if (ls_active == true || msta.Bits.RA_DONE || msta.Bits.RA_PROBLEM)
    {
        pmr->movn = 0;
        if (ls_active == true || msta.Bits.RA_PROBLEM)
        {
            clear_buttons(pmr);
            if (msta.Bits.RA_PROBLEM)
            {
                pmr->stop = 1;
                MARK(M_STOP);
            }
        }
    }

Because the XPS axes are normally in problem state before a home, issuing the home via the motor record also causes it to immediately send a stop, causing the home procedure on the XPS to fail. I'm not sure if this was part of the same record processing, or the next time it processes before the XPS driver has had the chance to run and un-set the problem bit.

I've submitted a pull request with a work around in the XPS driver. It clears the problem bit as part of the home function in the driver, which seems to work.

However, it seems a bit draconian to always do a stop if the record has the problem bit set. I'm sure in some cases that's what we want to do, but surely there are cases where the problem bit is set and a move is required to get out of problem state (whatever that problem state might be)?

On most of my XPS axes I have a custom home routine that masked this problem. My routine was putting the axes in and out of referencing mode, before issuing the real home, which seems to be more reliable. Only 1 of my XPS axes was still using the plain home function (and that was rarely used).

MarkRivers, 2016-04-05:
I have recently run into this problem.  I did not understand why I could not home my XPS axes any more from EPICS, when it used to work fine.  Now I understand.  There really needs to be more notification and testing when logic changes like this are made to the motor record.

tboegi, 2016-04-06:
A possible solution is to suppress the problem bit while homing:
https://github.com/epics-modules/motor/pull/26

mp49, 2016-04-06:
I'm guessing the feature was added because someone wanted to stop motion if the driver indicates there is a problem. In that case, it might be more appropriate to add that feature to the particular driver. Or, if it's application specific, it could be done at the database level by monitoring MSTA.

kmpeters, 2016-04-06:
This is the relevant commit: [303a9208e368ef07839afff67bac69e429af7332](https://github.com/epics-modules/motor/commit/303a9208e368ef07839afff67bac69e429af7332)

And here is comment from the source, to save a click:

> Set "stop" field true if driver returns RA_PROBLEM true. (Motor record stops motion when controller signals error but does not stop motion; e.g., maximum velocity exceeded.)

It seems to me that PROBLEM bit of the MSTA is not the correct bit to set when an axis is uninitialized; there is a HOMED bit which exists for this purpose.

mp49, 2016-04-07:
For the XPS controller the uninitialized state usually means there's a problem, because we can't move the axis until we re-home it (an annoying feature of the XPS). The problem bit is the only way we can indicate a problem to the high level software (if it's not a limit switch or a following error).

For the XPS home issue I've submitted a pull request with a fix to the driver, which works around this issue.

For other drivers that set the problem bit, I'm not sure if this new feature is a problem or not (which is why I opened this issue). I'm not sure we should always send a stop if the driver sets the problem bit.

For the Parker 6K driver it isn't an issue, because the only times I set the problem bit is when there is a amplifier fault, and so extra stops don't make any difference because the axis isn't going anywhere until we fix the amplifier.

kmpeters, 2016-04-07:
> For the XPS controller the uninitialized state usually means there's a problem, because we can't move the axis until we re-home it (an annoying feature of the XPS). The problem bit is the only way [...]

The motor documentation describes the PROBLEM bit this way:
`10. PROBLEM: driver stopped polling, or hardware problem`
And:
`The record is put into MAJOR STATE alarm if either SLIP_STALL or PROBLEM bits are detected.`

The XPS controller has ten status codes for "Not initialized states".  Codes one through nine correspond to states where setting the problem bit of the MSTA seems appropriate; the axis is uninitialized because something went wrong.  Code zero simply means the axis hasn't been homed yet.

> For the XPS home issue I've submitted a pull request with a fix to the driver, which works around this issue.

I merged that pull request.

> For other drivers that set the problem bit, I'm not sure if this new feature is a problem or not (which is why I opened this issue). I'm not sure we should always send a stop if the driver sets the problem bit.

It depends on what the problem bit means.  If the problem bit is used to signal a serious problem with the driver or the hardware, stopping the motor seems reasonable.   If the problem bit is used to signal any state where the a move won't succeed, then stopping the motor may not be reasonable.

I would like to see the problem bit only set when there is actually a serious problem.  Drivers read the status of motor controllers and can ignore move commands when the controller is in a state where moves should be disallowed.

kmpeters, 2018-05-02:
Worked around by #24

closed 2018-05-02 by kmpeters