epics-modules/motor #27: Motion allowed when DLLM > DVAL > DHLM

kmpeters, 2016-04-06:
### Setup
A motor, xxx:m1, has the following settings:
```
xxx.m1.DHLM = -10.0
xxx.m1.DVAL = -3.0
xxx.m1.DLLM = 10.0
```
### Expectation
* Moves in either direction will be disallowed and LVIO will be set to 1.

### Reality
* Moves are allowed as long as DLLM > DVAL > DHLM.
* LVIO is set to 1, but the move is not aborted:
  * during a backlash move when BDST != 0
  * during a move that was initiated while another move was in progress (regardless of the NTM value)

tboegi, 2017-02-13:
Does everybody agree on this ?
Some motion controllers treat LLM > HLM as "no soft limits at all, please"
The suggestion here says LLM > HLM as "no move at all, please".

=================
Another question:
Are these 2 related, but different problems:
Problem 1): The DLLM > DVAL > DHLM equation from above.
Problem 2): LVIO is set, but the motor is not stopped. (I think this is a bug)

kmpeters, 2017-02-13:
> Does everybody agree on this ?
> Some motion controllers treat LLM > HLM as "no soft limits at all, please"

Which motor controllers allow disabling soft limits by setting the LLM greater than the HLM?

> The suggestion here says LLM > HLM as "no move at all, please".

The motor record ignores soft-limits if DHLM = DLLM = 0.  In the scenario where LLM > HLM, every move from a starting point that is between the limits would violate either LLM or HLM.  "No move at all, please" is the only behavior that makes sense under those conditions.

> =================
> Another question:
> Are these 2 related, but different problems:
> Problem 1): The DLLM > DVAL > DHLM equation from above.
> Problem 2): LVIO is set, but the motor is not stopped. (I think this is a bug)

I believe the 2 issues are related, but I don't know if they can be separated.  If the first issue is fixed, the second issue may no longer exist.

I thought I used commit ae84076e42347a365b360d1d5f1583a07805b02d when I created the github issue, but the behavior I'm seeing right now is slightly different that problem 2.

Here is what I now see when I try to recreated problem 2:

**Scenario 1: Properly configured limits**
Setup
```
xxx:m1.DHLM = 10.0
xxx:m1.DVAL = 9.0
xxx:m1.DLLM = -10.0
xxx:m1.BDST = -0.5
```
Move
```
caput xxx:m1.DVAL 9.75
```
Result
```
Motor doesn't move
xxx:m1.LVIO = 1
```

**Scenario 2: Backwards limits**
Setup
```
xxx:m2.DHLM = -10.0
xxx:m2.DVAL = 9.0
xxx:m2.DLLM = 10.0
xxx:m2.BDST = -0.5
```
Move
```
caput xxx:m2.DVAL 9.75
```
Result
```
Motor moves to 10.25 at normal speed, then to 9.75 at backlash speed 
xxx:m2.LVIO = 0 (for the entirety of the move)
```

tboegi, 2017-02-14:
Excellent, thanks for the description.
I will make a test case out of that.

PR #108 (2018-08-21): Set LVIO=1 if DLLM > DHLM
Set LVIO=1 if DLLM > DHLM, since at least one limit being violated.

Fixes #27
closed 2018-08-21 by kmpeters