epics-modules/motor #41: executing forward link was made conditional.  This breaks databases that use motors

timmmooney, 2016-09-28:
When the motorRecord fails incorrectly to execute its forward link - for example, in response to a write to the .STOP field of an already stopped motor - the record is left in an altered state that can cause numerous subtle problems in databases that use the motor record.  Many databases in the optics module rely on EPICS' processing rules to break infinite loops, and those rules depend on whether the write came from channel access or database access.  (The rule: if a record is processing as the result of a database write, feedback processing resulting from that write that involves another database write to the record doesn't set the .RPRO field, which would cause the record to be processed again on completion.)  If a channel access write to .STOP doesn't execute the forward link, then the record's .PUTF field remains at 1, the next database put to any field will appear to have been made by channel access.  This breaks the optics 2slit_soft database.

MarkRivers, 2016-09-28:
What commit introduced this problem?

kmpeters, 2016-09-28:
Probably this one:

https://github.com/epics-modules/motor/commit/0ef39053aac72d8718ef99d08164da1be0825370

timmmooney, 2016-09-28:
This one (Feb 25, 2015):
- Fix for excessive FLNK processing reported by Alexander Söderqvist …

rsluiter, 2016-09-28:
I've submitted [this pull request](https://github.com/epics-modules/motor/pull/42) to address Tim's comment. As noted in the pull request, recGblFwdLink is always called when the motor record is processed unless DMOV is FALSE. Other than the DMOV condition, motor record calls to recGblFwdLink is standard.

tboegi, 2016-09-30:
Some general comment, not only for this PR:
Could the original description be part of the commit message?
That would make the commit self-contained.
People reading "git log" would immediately understand what is going on,
and especially, why.

tboegi, 2016-09-30:
Comment about the PR:
Does it make sense to make the "excessive forward link processing" runtime configurable ?
Some applications want it, other don't.

timmmooney, 2016-09-30:
I think it doesn't make sense.  If people want an output link that executes only when the motor actually moves, we could add a link that does that.  But changing the way the forward link works has side effects that are wide ranging and that require a lot of information to anticipate, because forward links are involved in so much of EPICS' management of record processing.  For example, if FLNK executes only when the motor actually has moved, then a put_callback that writes the current position to VAL will hang until some third party actually does move the motor.  Nobody wants side effects like this.

anjohnson, 2016-09-30:
Tim is right; the IOC's record processing completion mechanisms for `ca_put_callback()` notifications and for reprocessing the record when necessary are all implemented inside `recGblFwdLink()`. A record type _could_ copy that code into its own `process()` routine and split it up, but I wouldn't want to advise on what might break if one did that, nor can I promise that the code inside `recGblFwdLink()` won't change without notice in future releases of Base.

mp49, 2016-10-03:
If we need to trigger database processing when a motor actually moves, this can be accomplished by a calcout record which has a monitor on DMOV. It can write an output when it sees a 0->1 transition. I've used this in the past when the motor record behavior wasn't desirable.

rsluiter, 2016-10-04:
If there are no objections or more comment on this issue, I will merge my pull request.

rsluiter, 2016-10-19:
Resolved with https://github.com/epics-modules/motor/pull/42 merge.

closed 2016-10-19 by rsluiter