epics-modules/motor #54: missing content in documentation: motorDeviceDriver.html

prjemian, 2017-02-07:
This file ends with `The asynMotorAxis base class defines the following methods:` but no further text is provided.

View as HTML: http://htmlpreview.github.io/?https://raw.githubusercontent.com/epics-modules/motor/master/documentation/motorDeviceDriver.html

View as raw: https://raw.githubusercontent.com/epics-modules/motor/master/documentation/motorDeviceDriver.html

View as content: https://github.com/epics-modules/motor/blob/master/documentation/motorDeviceDriver.html

prjemian, 2017-02-07:
@MarkRivers: The commit d01040c7c left this section with missing content.  Should content be added or should the section be removed?

MarkRivers, 2017-02-07:
That document needs to be completed.  This is that last commit message:

Author: MarkRivers <rivers@cars.uchicago.edu>
Date:   Mon May 25 19:45:52 2015 +0000

    Removed old stuff about mar345 left over from document creation; document is still very incomplete

It would be very useful for this document to describe the Model 3 API, including the multi-axis coordinated move methods.

I'll try to make some time to work on it.

kmpeters, 2018-08-21:
I recently renamed the documentation directory so we could use the github.io url.  Updated links are below.

View as HTML: https://epics-modules.github.io/motor/motorDeviceDriver.html

View as raw: https://raw.githubusercontent.com/epics-modules/motor/master/docs/motorDeviceDriver.html

View as content: https://github.com/epics-modules/motor/blob/master/docs/motorDeviceDriver.html

kmpeters, 2018-08-21:
@MarkRivers, there will be a 6-11 release of the motor module in the near future.  Would you like the missing documentation to be included in it?  I realize that the missing documentation is very likely the least important thing on your to-do list.

MarkRivers, 2018-08-25:
I don't think I will have a chance to complete that document before the release.  I added a note at the end of the file saying it still needs to be completed to reduce confusion.

prjemian, 2019-03-06:
[R6-11](https://github.com/epics-modules/motor/releases/tag/R6-11) has been released.  We should move this issue to another milestone **and** close the [R6-11](https://github.com/epics-modules/motor/milestone/4) milestone.