epics-modules/motor #58: motorApp/MotorSrc/README is a guess

jlmuir, 2017-03-07:
[`motorApp/MotorSrc/README`](https://github.com/epics-modules/motor/blob/4f54eb502bb1e4157db9a0587cfdbabd66de0c95/motorApp/MotorSrc/README) begins with the following line:

```
Here's my best guess at what all this stuff is. TMM
```

I really appreciate the effort to create a best guess, but an expert should confirm, and then that line should be removed.

MarkRivers, 2017-03-07:
I just updated the README so I believe it is now accurate.

MarkRivers, 2017-03-07:
These are the differences:
```
diff --git a/motorApp/MotorSrc/README b/motorApp/MotorSrc/README
index 9978185..f4eabee 100644
--- a/motorApp/MotorSrc/README
+++ b/motorApp/MotorSrc/README
@@ -1,5 +1,3 @@
-Here's my best guess at what all this stuff is. TMM
-
 Model 3 (asyn, C++) driver base classes
 --------------------------------
 asynMotorAxis.cpp
@@ -10,17 +8,20 @@ asynMotorController.h
 Model 2 and Model 3 device support
 --------------------------------
 devMotorAsyn.c
+motor_interface.h

 Model 2 (asyn, C)
 --------------
 drvMotorAsyn.c
-motor_interface.h
 paramLib.c
 paramLib.h

+Definitions used in record, device support, and utilities
+---------------------------------------------------------
+motor.h
+
 motor record
 ------------
-motor.h
 motorRecord.cc
 motorRecord.dbd
```

closed 2017-03-08 by jlmuir