epics-modules/motor #61: Motor record can lock up when setting DVAL in SET mode

mp49, 2017-04-04:
I've come across an edge case problem which causes the motor record to lock up (which is fixable by toggling SPMG). When using the Galil model 3 driver I can get the record to lock up (DMOV stays 0) by first setting the offset (OFF), then switching to SET mode, then setting DVAL.

The motor record sets STUP=BUSY when setting the OFF field, and issues a GET_INFO command to device support. However, if the driver doesn't cause a callback to device support then STUP will stay BUSY. With the Asyn drivers we normally get a callback because the motorUpdateStatus_ is handled like this in asynMotorController::writeInt32:

if (function == motorUpdateStatus_) {
    bool moving;
    /* Do a poll, and then force a callback */
    poll();
    status = pAxis->poll(&moving);
    pAxis->statusChanged_ = 1;

which forces a callback to device support even if nothing has changed. This will cause the record to process and to clear the STUP field. Then the subsequent process due to a SET+DVAL will successfully set back DMOV.

However, the Galil driver implements it's own motorUpdateStatus_ function in it's derived version of writeInt32. This doesn't result in a callback if nothing has changed. This means the motor record stays with STUP=BUSY even when moving on to the SET+DVAL function. 

So there are two solutions:

1) Callbacks are required by the motor record when doing STUP or GET_INFO. Then we fix this in the Galil driver by doing something like:

//Force a callback even if nothing has changed.
pAxis->setIntegerParam(this->motorStatusDone_, 0);
pAxis->setIntegerParam(this->motorStatusDone_, 1);
pAxis->callParamCallbacks();

This would be added to the GalilController::writeInt32 function in the section that deals with motorUpdateStatus_. Toggling motorStatusDone_ will force a callback.

2) Reset STUP when beginning the set position routine in the motor record, like:

pmr->stup = motorSTUP_OFF;

which would be added to the load_pos() function in the motor record.

I've tested both solutions and they fix the issue. I'm not sure which is the right solution though.
I believe this only affects the Galil model 3 driver because I think most others (or all others) make use of the default version of the motorUpdateStatus_ code in asynMotorController.

tboegi, 2017-04-05:
To improve the Galil driver seems to be a good idea, since
the driver should to a callback even if nothing is changed even for other cases:

http://www.aps.anl.gov/epics/tech-talk/2015/msg01786.php

And it seems as if a simple
pAxis->statusChanged_ = 1;
would do:

  //GalilPoller is designed to be a constant and fast poller
  //So we dont honour motorUpdateStatus as it will cause threading issues
  if (function == motorUpdateStatus_)
	{
	//Dont block synchronous poller
	unlock();
	//Give poller time to provide status update to mr, then we're done
	epicsThreadSleep(updatePeriod_*2.0/1000.0);
	lock();
        pAxis->statusChanged_ = 1; /* Must force a callback */
	return asynSuccess;

mp49, 2017-04-05:
Exiting SET mode still leaves DMOV=0. 
OFF can be set either before entering SET mode of after (but before setting DVAL).
I tried SPMG toggle in SET mode and in USE mode and it fixes it in both cases.

mp49, 2017-04-05:
I tried:
pAxis->statusChanged_ = 1;
in the Galil writeInt32 function, which I thought would have worked except it didn't because it's a private data member of asynMotorAxis. I guess we could easily add a function to allow that to be set via derived classes, or make it protected.

mp49, 2017-05-02:
This was fixed in the Galil driver.

closed 2017-05-02 by mp49