epics-modules/motor #76: Incorrect acceleration calculation for controllers that don't support VBAS

kmpeters, 2017-08-28:
### Problem

If a motor controller doesn't support setting a base velocity and the associated motor record is configured to have a non-zero VBAS, the motor record will calculate an acceleration that results in a shorter acceleration period than the one specified by ACCL.

### Proposed solution

- Add a bit to the MSTA field to allow drivers to tell the motor record if they support setting a base velocity (0=base velocity is supported, for backwards compatibility).
- Set this new bit to 1 in every driver that doesn't support setting a base velocity.

### Previous discussion

https://github.com/epics-modules/motor/pull/75

tboegi, 2017-09-04:
A first attempt is here:
https://github.com/epics-modules/motor/pull/80

kmpeters, 2018-05-02:
Likely to be fixed by #81

kmpeters, 2018-07-11:
#81 was closed, so this will still be an issue.

An example of how drivers can workaround the problem by correcting the acceleration passed by the motor record to give the desired acceleration time (ACCL) can be found here:

https://github.com/epics-modules/motor/blob/3683d3f0eaae1b5893cd32615174b646cf259581/motorApp/AMCISrc/ANF2Driver.cpp#L632

tboegi, 2018-07-12:
>#81 was closed, so this will still be an issue.
It was closed because
a) Adding bits to MSTA is not a good way to go
b) The problem is that the driver may want to "push" VBAS from the driver into the record,
    and this is not at all addressed by the patch.

A possible solution how to push values from driver (in this case the softlimits,
sometimes named "read only soft limits) can be seen here:
https://github.com/EuropeanSpallationSource/motor/blob/ess-6.9.x/motorApp/MotorSrc/motorRecord.cc#L1525