epics-modules/motor #82: Encoder ratio sign is always positive when set via motor record

mp49, 2017-09-27:
If we have MRES and ERES with opposite signs the encoder ratio can be set incorrectly and the sign information lost. 

In the asyn device support this is done correctly at IOC startup in the init_controller function. But if we then change MRES or ERES then the motor record issues a SET_ENC_RATIO with these values:

ep_mp[0] = fabs(m / pmr->eres);
ep_mp[1] = fabs(m / pmr->mres);

and the sign information can be lost. The motor record matches the non-asyn device support though (see the motor_init_record_com function in motordevCom.cc).

Removing the use of fabs should fix it.

Is opposite sign MRES and ERES meant to be allowed?

rsluiter, 2017-09-27:
Chapter and verse from https://www3.aps.anl.gov/bcda/synApps/motor/R6-9/motorRecord.html#Fields, under MRES;
"MRES or UREV allow negative values so that the user/dial coordinate systems can be configured to the opposite polarity of the motor controller's."
So yes, the motor record supports MRES and ERES of opposite signs. My assumption was that motor controllers would only accept the absolute value of the ratio. I can't imagine why the sign of the ratio would matter. Was my assumption wrong?

mp49, 2017-09-27:
Hi Ron,

It seems to be an issue when doing a 'set position'. The asyn drivers use the encoder ratio to figure out what to set for the encoder position (by multiplying the motor position passed down into the driver). So if the sign is wrong on the encoder ratio then the encoder position written to the controller will be the wrong sign.

mp49, 2017-09-27:
I think this could be fixed at the Asyn device support level too, in this section:

case SET_ENC_RATIO:
        pmsg->command = motorEncRatio;
        pmsg->dvalue = param[0]/param[1];
        break;

Since we still have access to the motor record fields we could calculate the correct sign.

I've confirmed it's an issue with two different Asyn drivers (the Parker 6K one and the Galil one), and I checked that removing the use of the 'fabs' function in the motor record fixes the problem for those two drivers.

I'm not sure if any drivers use the encoder ratio in a different way though, so it's hard to know if a change like this would break anything.

rsluiter, 2017-09-27:
This could be fixed by removing the "fabs" from the code in question and dealing with the encoder ratio polarity in the drivers. The OMS VME58 and the Newport MM3000 controllers are the only ones, that I can find, that use the encoder ratio. That narrows the scope of work.

rsluiter, 2017-09-27:
I have no preference. Either fix should work.
With regard to "it's hard to know if a change like this would break anything."; this is especially true of with motor drivers outside the EPICS motor module.

kmpeters, 2018-04-27:
@rsluiter should #84 be merged now?

kmpeters, 2018-07-03:
This was fixed with #84.

closed 2018-07-03 by kmpeters