epics-modules/motor #85: Problem with R6-10 issue 6 fix - controller position can be lost on reboot

mp49, 2017-10-03:
Hi,

I've come across a problem with the change documented by R6-10 issue 6 (the use of use_rel in devMotorAsyn.c in the init_controller). There's a few cases where this can cause us to lose the controller position:

1) Stop the IOC, then the motor is moved using a 3rd party program, then start the IOC. The IOC will restore the incorrect autosave position and we lose the controller position.
2) Move the motor, restart the IOC while the motor is moving, the IOC restores the previous autosaved position (which may or may not be accepted by the controller while it's moving).
3) Move the motor a short distance and wait until it stops, restart the IOC before autosave has chance to save the position, then the IOC will restore the previous incorrect position.

I understand the reasoning behind the change, but I think a better way would be to automatically sync the motor position to the encoder position at the end of a move, providing UEIP=Yes and RTRY is non-zero. 

I'm having to roll back to 6-9 because of this change because there's too many situations where it could cause problems.

rsluiter, 2017-10-03:
To be clear; you are referring to the pre-github [<top>/README ](https://github.com/epics-modules/motor/blob/master/README) file's R6-10 item #6, which begins with "Kevin Peterson discovered an initialization bug when the motor record is configured to do retries."?

mp49, 2017-10-03:
Yes, that's the one.

tboegi, 2017-10-09:
Does anybody, which commit ID this is ?

ulrikpedersen, 2017-10-10:
> Does anybody know, which commit ID this is ?

When looking at the README with the (really neat) github 'blame' feature switched on you can see the commit. Fortunately that particular commit to the top/README that @rsluiter refers to also include the C code. It seems the work was done in 3090983c313c57efdcd7c788ba234e775d300e06 presumably based on findings from @kmpeters

tboegi, 2017-10-19:
That opens 2 questions:
@rsluiter , @kmpeters : For which controller was the change made ?
Does this controller report "not homed" after reboot and "homed" after setPostion ?

Same question @mp49:
Does the parker6K switch from "not homed" to "homed" ?

mp49, 2017-10-19:
For the parker6k driver I only set the homed bit to 1 if the controller reports 'home successful' after a home sequence on the controller (either home to home switch or reference mark).

We've come across the same issue described in the 3090983 commit many times. One problem is that the motor position can gain a large offset compared to the encoder position. On the Parker 6K the soft limits apply to the motor position and not the encoder position, and so we were running into soft limits even though the encoder position was within range. We solved this by either setting the soft limits very wide or disabling soft limits, but I think the best solution is to sync the motor position to the encoder position (if RTRY!=0 and UEIP=1). The model 3 Galil driver already does this, and I will probably add the same feature to the parker6k driver. I guess it could be a new motor record feature too (optional, enabled via a new field).

One small advantage of letting the motor position get out of sync over time is that it's really obvious that there's an issue with the system. We archive both the motor position and encoder position and so we can track if they diverge over time.

kmpeters, 2017-10-19:
I don't have time at the moment to get into the details of this issue, but I want to share some of the history of the https://github.com/epics-modules/motor/commit/3090983c313c57efdcd7c788ba234e775d300e06 commit, which was made before the transition to github.  These are my notes on the problem:
```
Subject
-------
External-readback initialization problem

Description
-----------
OMS VME58 motor records with the following configuration do not initialize properly:

1. DTYP = "OMS VME58"
2. URIP = "Yes"
3. RRES is non-zero (RRES=1 if the RDBL PV has the same units as EGU)
4. RDBL contains a link to a PV with the desired readback value
5. RMP != RDBL PV converted into raw units

The restored position is consistent with the RMP, but the motor record should have a position matching that of the RDBL PV.

How to recreate the problem:

1. Ensure that the VME58 motor's position and settings are autosaved
2. Configure the VME58 motor to use an external readback
a. URIP=Yes
b. RRES>0
c. RDBL has link to PV with a valid readback value
2. Create a discrepancy between the RMP and the RDBL PV
a. Set RTRY to be a moderately-large, non-zero number
b. Initiate a move, preferably with a size of 1 EGU, but don't allow the RDBL PV to change.
c. Wait for the move to complete. Each retry will increase the position discrepancy by the size of original move request.
3. Confirm that the autosaved position and settings for the motor are up-to-date.
4. Soft-reboot the vme (ctrl+x or 'reboot')

Recovering from the problem:

Requesting a status-update (setting the STUP field to 1) will update the RBV and DRBV so that they are consistent with the RDBL PV. The VAL, DVAL, and RVAL will still be incorrect (consistent with the RMP). The SYNC field does NOT resolve the issue. Setting the VAL field to the RBV is required to synchronize the VAL fields with the RBV fields. If the RMP results in a limit error at startup, the limit error will not be cleared until a move is attempted. The RRBV will continue to mirror the RMP. The only way to synchronize it with the RDBL PV is to (1) power-cycle the vme or (2) set the DVAL to the current position.
```

These are some of Ron's comments on the issue:
```
        After Kevin demonstrated the problem raised by this issue, it was clear that the root cause of the problem is that the motor record initialization logic that determines how the target positions are initialized at boot-up is not taking into consideration whether or not the motor record is configured to do retries, and therefore, whether or not absolute or incremental moves will be issued by the motor record.

        If retries are active and incremental moves were being used when the IOC was shutdown, then at boot-up, the absolute position from the controller should be ignored (because the controller's absolute position was "corrupted" by retries) and the autosave/restore position should be used to set the controllers position.

        Added "use_rel" indicator to motor_init_record_com() in motordevCom.cc and to init_controller() in devMotorAsyn.c. Added the "use_rel" indicator to the "initPos" logic in both motor_init_record_com() and init_controller() . 
```

mp49, 2017-10-19:
That sounds like a different issue from the one I thought the patch was solving. 

Maybe we just need a way to be able to set the readback RBV position (external PV or actual encoder) at IOC start? Which would mean autosaving RBV (but not sure we can use autosave for that).

mp49, 2018-08-21:
Thanks, yes I think that will work.

Blasater, 2018-08-21:
Sorry to jump in here off topic.  Anyone in here know, or know of, someone with Newport 500C motion controller knowledge?  

Thanks!

Brian

From: Matt Pearson 
Sent: Tuesday, August 21, 2018 9:34 AM
To: epics-modules/motor 
Cc: Subscribed 
Subject: Re: [epics-modules/motor] Problem with R6-10 issue 6 fix - controller position can be lost on reboot (#85)

Thanks, yes I think that will work.

anjohnson, 2018-08-21:
@kmpeters I hope your new field would actually be a record-specific menu, with suitable choice strings to make the two behaviors obvious to humans reading them...

kmpeters, 2018-08-21:
@Blasater, you should send your question to tech-talk: https://epics.anl.gov/tech-talk/

FreddieAkeroyd, 2019-10-30:
Hi, I was just wondering what the status of this was as this issue will currently stop us updating our motor record. I like the idea of a new motor record field to govern behaviour, has there been any further discussion on the field name and potential values? We might have some time to work on this, I just wanted to check if anybody else was currently (or planned to start) working on it.

kmpeters, 2019-10-31:
As far as I know there hasn't been any further discussion about the field name and potential values.  I haven't started working on it yet.  Now that the motor split is complete I'm happy to raise the priority of this issue.  I'll create a dedicated issue for it now.

kmpeters, 2023-05-15:
Fixed by #160