epics-modules/motor #91: EPICS 7 and caput crashes IOC

prjemian, 2018-01-05:
* arch: linux-x86_64
* base: 7.0.1.1
* synApps: 5.8
* motor: master, commit hash 60aa414b37
* xxx: 5.8.4
* problem: IOC crashes when client tries to move a motor (soft motor xxx:m1.VAL)
* may be related to: https://github.com/epics-modules/xxx/issues/8

gdb session:

```
username@host ~/.../iocBoot/iocLinux $ gdb ../../bin/linux-x86_64/xxx 
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ../../bin/linux-x86_64/xxx...done.
(gdb) run st.cmd
Starting program: /usr/local/epics/synApps/support/xxx-R5-8-4/bin/linux-x86_64/xxx st.cmd
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff08c7700 (LWP 17343)]
[Thread 0x7ffff08c7700 (LWP 17343) exited]
# Linux startup script
##########################  ############################################################################
# environment variable      description
##########################  ############################################################################
# PREFIX		    prefix used for all PVs in this IOC
# EPICS_DB_INCLUDE_PATH     search path for database files
# EPICS_CA_MAX_ARRAY_BYTES  Specify largest array CA will transport
# IOCBOOT		    absolute directory path in which this file is located
# RHOST                     specifies the IP address that receives the alive record heartbeats
##########################  ############################################################################
epicsEnvSet("PREFIX", "xxx:")
### For alive
epicsEnvSet("RHOST", "1.2.3.4")
[New Thread 0x7ffff7faa700 (LWP 17344)]
macLib: macro TOP is undefined (expanding string epicsEnvSet("IOCBOOT", $(TOP)/iocBoot/$(IOC)))
macLib: macro IOC is undefined (expanding string epicsEnvSet("IOCBOOT", $(TOP)/iocBoot/$(IOC)))
### For devIocStats
epicsEnvSet("ENGINEER","engineer")
epicsEnvSet("LOCATION","location")
epicsEnvSet("GROUP","group")
macLib: macro DEVIOCSTATS is undefined (expanding string epicsEnvSet("EPICS_DB_INCLUDE_PATH", "$(DEVIOCSTATS)/db"))
< envPaths
epicsEnvSet("IOC","iocLinux")
epicsEnvSet("TOP","/usr/local/epics/synApps/support/xxx-R5-8-4")
epicsEnvSet("SUPPORT","/usr/local/epics/synApps/support")
epicsEnvSet("EPICS_BASE","/usr/local/epics/base-7.0.1.1")
epicsEnvSet("ASYN","/usr/local/epics/synApps/support/asyn-R4-32")
epicsEnvSet("AUTOSAVE","/usr/local/epics/synApps/support/autosave-R5-9")
epicsEnvSet("BUSY","/usr/local/epics/synApps/support/busy-R1-7")
epicsEnvSet("CALC","/usr/local/epics/synApps/support/calc-R3-7")
epicsEnvSet("CAPUTRECORDER","/usr/local/epics/synApps/support/caputRecorder-R1-7")
epicsEnvSet("DEVIOCSTATS","/usr/local/epics/synApps/support/iocStats-3-1-15")
epicsEnvSet("IP","/usr/local/epics/synApps/support/ip-R2-19")
epicsEnvSet("IPAC","/usr/local/epics/synApps/support/ipac-2-14")
epicsEnvSet("MCA","/usr/local/epics/synApps/support/mca-R7-7")
epicsEnvSet("MODBUS","/usr/local/epics/synApps/support/modbus-R2-10")
epicsEnvSet("MOTOR","/usr/local/epics/synApps/support/motor")
[... 648 lines cut ...]
    at strncat_chk.c:50
50	strncat_chk.c: No such file or directory.
```

stack trace after error

```
(gdb) bt
#0  __strncat_chk (s1=0x7fff7ae1c9e6 ".", s1@entry=0x7fff7ae1c9e0 "xxx:m1.", s2=0x0, n=n@entry=99, s1len=92, s1len@entry=100)
    at strncat_chk.c:50
#1  0x00007ffff2d9f4c4 in strncat (__len=99, __src=<optimized out>, __dest=0x7fff7ae1c9e0 "xxx:m1.")
    at /usr/include/x86_64-linux-gnu/bits/string3.h:156
#2  my_dbNameOfPV (bufLen=100, paddr=0x7fff8c25ec48, paddr=0x7fff8c25ec48, pBuf=0x7fff7ae1c9e0 "xxx:m1.")
    at ../caputRecorder.c:266
#3  myAsDataListener (pmessage=0x7fffa00019b8, after=<optimized out>) at ../caputRecorder.c:309
#4  0x00007ffff266fa6b in asTrapWriteAfterWrite (pvt=pvt@entry=0x7fffa00019a8) at ../../src/as/asTrapWrite.c:167
#5  0x00007ffff292626a in write_action (mp=0x7fff7ae1cd70, pPayload=<optimized out>, client=0x7fff94000f88)
    at ../../../src/ioc/rsrv/camessage.c:805
#6  0x00007ffff2924abd in camessage (client=client@entry=0x7fff94000f88) at ../../../src/ioc/rsrv/camessage.c:2538
#7  0x00007ffff2921811 in camsgtask (pParm=0x7fff94000f88) at ../../../src/ioc/rsrv/camsgtask.c:117
#8  0x00007ffff268c65c in start_routine (arg=0x7fffa0000fd0) at ../../src/osi/os/posix/osdThread.c:403
#9  0x00007ffff134e6ba in start_thread (arg=0x7fff7ae1d700) at pthread_create.c:333
#10 0x00007ffff23873dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
(gdb) 
```

MarkRivers, 2018-01-05:
That appears to be the same error in caputRecorder that I posted a patch for in https://github.com/epics-modules/xxx/issues/8.  Tim says he has committed a new version of caputRecorder with that fix.  Are you using his fix?  If not please try it.

prjemian, 2018-01-05:
rebuilt synApps (`make rebuild`) with commit c1c440764414

confirmed the commit in caputRecorder (hash: 256f57fcc) resolves the error

closed 2018-01-05 by prjemian