epics-modules/motor #92: Change "Offset-Freeze Switch" field (FOFF) from "Variable" to "Frozen".

rsluiter, 2018-01-10:
The only "use case" I know of for the FOFF field to be "Variable" is if the user wants the OFF field to be automatically calculated when the motor record's Set/Use Switch is in the "Set" state and a new DVAL is entered. Under this scenario, the motor record will set,
OFF = VAL - (DVAL * DIR) 
Otherwise, I find that FOFF = "Variable" causes user confusion. Therefore, I would like to change the FOFF default to "Frozen".  Comments or objections?

MarkRivers, 2018-01-10:
That use case is what we want 90% of the time.  We set our motors so that the dial position is referenced to home or the negative limit switch.  When the user sets the VAL field we don't want it to change the DVAL, we want it to set the OFF.

timmmooney, 2018-01-10:
In many cases, dial coordinates are precious, and beamline staff don't want to risk their loss, but users want to recalibrate the current position.  FOFF="Variable" serves that common need.  FOFF="Frozen" is a risky setting, in my view.  This is particularly true for databases that use the motor record's Set/Use switch to implement a higher level calibration.

rsluiter, 2018-01-11:
OK. I will drop this. Thanks for the feedback.

closed 2018-01-11 by rsluiter