epics-modules/motor #106: Acceleration always changes with VELO

kmpeters, 2018-08-13:
The ACCL field of the motor record is an acceleration time with units of seconds.  Changing the ACCL field changes the acceleration, which is obvious.  Changing the VELO field also changes the acceleration, which is not always obvious.  The motor record should have a way to keep the acceleration constant when changing the VELO, which should be optional and off by default for backwards compatibility.

MarkRivers, 2018-08-13:
One nice way to handle this might be the following:
- Add a new acceleration field whose units are egu/s/s.   Perhaps the name would be ACSS (acceleration in s/s).  
- If ACCL is written to then:
  - Current method of acceleration computation is used:
  - ACCS is computed and posted
- If ACSS is written to then:
  - Acceleration is computed using ACSS, which more closely matches the value written to the driver
  - ACCL is computed and posted

The motor record keeps track of whether ACCL or ACCS was last written to.

This should be backwards compatible but allows controlling acceleration in the more logical units, which also closely matches the units passed to the driver (the difference being just MRES, i.e. difference between EGU and steps).

kmpeters, 2018-08-13:
I like that approach, but which acceleration field should change if VELO changes?  I think it probably makes sense to add another field to select whether ACCL or ACSS should be held constant.

MarkRivers, 2018-08-13:
My thought is that users who want the new behavior write to the ACCS field.  Users who write to the VELO field get the existing behavior.  The motor record keeps track of which field was last written to.  Changing VELO will change ACCL if ACCS was last written to, and will change ACCS if ACCL was last written to.

kmpeters, 2018-08-13:
I understand now.  One potential downside to this approach is that it isn't possible for a user to predict which acceleration field will change if that user wasn't the last person to move the motor.

Adding a field to select whether ACCL or ACSS should be held constant when VELO changes would make more complicated for users who regularly want to switch between the two (when compared to the approach you suggest), but it would make the motor record behavior more predictable in situations where multiple people/clients are changing motor settings.

tboegi, 2018-11-06:
Just to understand the problem:
Are there applications that simply control VELO but not ACCL?
People from other Neutron science facility say that the user of a motor
seldom cares about the acceleration and rather trust the engineers to configure
it to the right values.
With that background, I was tempted to suggest a following change:
Allow ACCL to be 0.0.
If now a movement is started, fall back to JAR (which is e.g. mm/sec2)
If the JAR field is 0.0 as well, fall back to the controller defaults.
This would need some work to be done in motorRecord.cc and possibly the driver(s).
Does this make sense to anybody ?

kmpeters, 2018-11-06:
My expectation is that most users will change VELO but not ACCL.  Changing the velocity by a factor of two results in a very noticeable reduction in the amount of time users need to wait for moves to complete.  Changing ACCL by a factor of two has a much more subtle effect.  This has been a problem for motor controllers that have maximum acceleration parameters, which can be easily exceeded by changing on the VELO in the motor record.

I don't think allowing ACCL and JAR to be zero is a good idea.  I think we should be reducing, not increasing the number of instances where the settings in the motor record are inconsistent with the behavior of an axis.

mp49, 2018-11-06:
I agree that ACCL is sometimes confusing to users, as most people are used to specifying acceleration in terms of egu/s/s. 

At our site I would say that 99% of motors don't get their motion parameters changed after commissioning. For the few motors that do have different speeds then I tend to provide a mbbo menu option next to the motor widget that sets VELO for the users, rather than have them drill down into the detailed motor screens. If we had to that could change ACCL too.

Although we do have expert users who know how to get to the VELO and ACCL fields and set them. Usually they are happy once I've explained that ACCL is acceleration time rather than acceleration.

How about having a new field which changes the behaviour of ACCL, rather than having a different acceleration field. The new field can be either set to 'Acceleration Time' (the default) or 'Acceleration'. If its set to 'Acceleration Time' then the current behaviour stays the same, and acceleration changes with VELO. If its set to 'Acceleration' then users will specify ACCL in terms of egu/s/s and it won't change when VELO changes. It does mean that ACCL will have to be recalculated if someone changes the new field.

tboegi, 2018-11-08:
In my last talks to motion control people it turned out that the acceleration should not
be user-changable at all, once the axis is commissioned by the engineer.
We can fix that in the driver, simply ignore the acceleration from the record.
That all is dependent on the motor/controller/mechanics.

Back to the comment
>This has been a problem for motor controllers that have maximum acceleration parameters
I am somewhat tempted to suggest an AMAX field, which defines the maximum acceleration,
measured in EGU/sec2.

timmmooney, 2018-11-08:
I would not be ok with making the acceleration off limits to users.  There are devices that cannot be fully commissioned without beam, because the precision required is at the limit of the device's capability, and users have to tweak things to get the best performance.

Tim Mooney (mooney@anl.gov) (630)252-5417
Beamline Controls Group (www.aps.anl.gov)
Advanced Photon Source, Argonne National Lab

MarkRivers, 2018-11-08:
I also think it is important to be able to change the acceleration.  If I have a sample translation stage and I change the load on it from 0.1 kg to 20 kg I may well have to change the acceleration.

tboegi, 2018-12-04:
I added the code as suggest (at least what I understand the suggestions):

https://github.com/epics-modules/motor/pull/122

And there is a possible test here:
<https://bitbucket.org/europeanspallationsource/m-epics-ethercatmc/commits/49cf67fa5d5faaffdaa5ab6592a325c039336a3e?at=181204_1052_test_for_ACCS_TC410>

Comments are welcome.

tboegi, 2019-03-06:
Please see 
https://github.com/epics-modules/motor/pull/122

kmpeters, 2019-03-06:
I would like to put this issue on hold until after the motor module is split into many vendor modules, since there won't be another major release before then and this would be a non-trivial change.

kmpeters, 2023-05-17:
@tboegi, yes I will test them locally.

kmpeters, 2023-05-17:
> I would be more comfortable with the following:
> 
>    * The `ACCU` field should be an "Acceleration Update" field that is set by the user (or database or autosave) and never automatically updated.
>    * The `ACCU` field should have two options: `Change ACCS` and `Change ACCL`.
>    * The `ACCU` field should only affect what happens when the `VELO` is changed.
>    * The `ACCU` field should have a default value of `Change ACCS` and be overridable when the db is loaded.
> 
> Changing `ACCS` would cause `ACCL` to update and vice versa, which is already the behavior in #122.

My proposal to invert the meaning of the ACCU field  ( `Change ACCS` instead of `Use ACCL` and vice versa) was not a good suggestion.  I think the original meaning is less confusing.  I'll make it clear in the documentation that the acceleration field that is used will remain fixed until the user changes it and the other field will change with the velocity.

I still think the ACCU field should be a control rather than a readback.  I'll push the branch when I resolve the initialization problems.

kmpeters, 2023-05-17:
I'll probably have to create a new motor release before I have more time to work on this issue.

tboegi, 2023-05-18:
@kmpeters 
I think that you simply can remove these 2 blocks:
But as I don't have a system to test (all our controllers define the ACCS inside the motion controller,
and not in the db file) I want to wait for more feedback.
```

--- a/motorApp/MotorSrc/motorRecord.cc
+++ b/motorApp/MotorSrc/motorRecord.cc
@@ -639,11 +639,6 @@ static double accEGUfromVelo(motorRecord *pmr, double veloEGU)
 
 static void updateACCLfromACCS(motorRecord *pmr)
 {
-    if (pmr->accu != motorACCSused_Accs)
-    {
-        pmr->accu = motorACCSused_Accs;
-        db_post_events(pmr, &pmr->accu, DBE_VAL_LOG);
-    }
     if (pmr->accs > 0.0)
     {
         double temp_dbl = pmr->velo / pmr->accs;
@@ -658,11 +653,6 @@ static void updateACCLfromACCS(motorRecord *pmr)
 static void updateACCSfromACCL(motorRecord *pmr)
 {
     double temp_dbl;
-    if (pmr->accu != motorACCSused_Accl)
-    {
-        pmr->accu = motorACCSused_Accl;
-        db_post_events(pmr, &pmr->accu, DBE_VAL_LOG);
-    }

```

kmpeters, 2023-05-18:
@tboegi, it was a little more complicated than that, but I got it working.  These are the changes I made to the branch used for #122:

https://github.com/EuropeanSpallationSource/motor/compare/torsten/181203_1217_add_ACCS_field...epics-modules:motor:add_accs_field_with_accu_control

My changes are in #203.  One significant change I made is to include VBAS in the acceleration calculations.

If everying looks OK, I'll merge #122 and then #203.

kmpeters, 2023-05-19:
I don't like that I removed these lines:

https://github.com/EuropeanSpallationSource/motor/compare/torsten/181203_1217_add_ACCS_field...epics-modules:motor:add_accs_field_with_accu_control#diff-57a971e1f43acd90092fb31fbfae954292d41aabb49705eba96b6239c47f06ecL3987-L3994

ACCL will be non-zero because it's required by the motor databases and autosaved by default.  ACCS will be zero unless someone takes action to make that happen (either by adding it to a motor database or autosave).  If they do that, the ACCS value is likely the important one.  I think the logic should probably look like this:

```
    if (pmr->accs > 0.0)
    {
        updateACCLfromACCS(pmr);
    }
    else if (pmr->accl == 0.0)
    {
        pmr->accl = 0.1;
        db_post_events(pmr, &pmr->accl, DBE_VAL_LOG);
    }
    else
    {
        updateACCSfromACCL(pmr);
    }
```

tboegi, 2023-05-19:
Don't we miss an update of ACCS here, When both ACCS == 0.0 and ACCL == 0.0 ?

` if (pmr->accs > 0.0)
    {
        updateACCLfromACCS(pmr);
    }
    else if (pmr->accl == 0.0)
    {
        pmr->accl = 0.1;
        updateACCSfromACCL(pmr);
        db_post_events(pmr, &pmr->accl, DBE_VAL_LOG);
    }
    else
    {
        updateACCSfromACCL(pmr);
    }`

kmpeters, 2023-05-19:
No.  If both are zero `pmr->accl` gets set to `0.1` and `updateACCSfromACCL` is called.

kmpeters, 2023-05-19:
It might be less confusing to write the code like this:
```
    if (pmr->accs > 0.0)
    {
        updateACCLfromACCS(pmr);
    }
    else
    {
        if (pmr->accl == 0.0)
        {
            pmr->accl = 0.1;
            db_post_events(pmr, &pmr->accl, DBE_VAL_LOG);
        }
        updateACCSfromACCL(pmr);
    }`
```

tboegi, 2023-05-19:
That's even better

kmpeters, 2023-05-19:
I'm happy with that initialization logic.  I also had to make `ACCL` or `ACCS` update when `VBAS` or `SBAS` changes:  https://github.com/epics-modules/motor/pull/203/commits/7291b5569be4084831fca363673acb5a9d59ac2b

The only way I've found to break the synchronization of `ACCL` and `ACCS` is to autosave and restore both of them, which results in them being temporarily out of sync until an acceleration or velocity field is changed.  This is something that is also broken for other motor fields that are synced like `MRES`, `SREV`, & `UREV` and the solution has been to not autosave all of them.

kmpeters, 2023-05-19:
I'll merge #122 and #203 on Monday if no one objects before then.

kmpeters, 2023-05-22:
Fixed by #122 and #203

closed 2023-05-22 by kmpeters