epics-modules/motor #110: devSoftMotor does not work properly with base 7.0

MarkRivers, 2018-09-08:
The symptoms included CA SEVCHK errors because of null PV pointers, failure to register CA callbacks for link changes, etc.

The problems have been found:
 - It was using the value of mr->[link]->value.constantStr and mr->[link]->value.pv_link.pvname without checking the link type.  This was incorrect, and led to errors because it appears that previously the link->constantStr contained the PV string even if the link->type was not CONSTANT.  This is no longer the case, the link->constantStr appears to be random if the link type is not CONSTANT.  
- It was not locking the motor record when accessing the record fields.  This is incorrect and dangerous.
- It did not check if pv_link.pvname was NULL before calling CA functions, which led to SEVCHK errors if the link was not defined.

MarkRivers, 2018-09-08:
Fixed via ee9fca1dc3b52d0599b7ff34afb53dcf032f5b16.  This is the commit entry:
```
    Fixes for problems that showed up in base 7.0.
    - It was using the value of mr->[link]->value.constantStr and mr->[link]->value.pv_link.pvname without checking the link types.
      This was incorrect, and led to errors because constantStr no longer has a defined value if the link type is not CONSTANT.
      Changed so that it only accesses these union fields if the link is the correct type.
    - It did not check if pv_link.pvname was NULL before calling CA functions, though now that we check the link type perhaps this cannot occur.
    - It did not lock the motor record before accessing the fields.
    - Added Debug calls to show the link types and values.
```

This fix has been tested to fix the problems on 7.0.1, and also still works on 3.15.5.  It should be looked at by others to see if more changes are needed.

tboegi, 2018-09-21:
Is this fixed with ee9fca1dc3b52d0599b7ff34afb53dcf032f5b16 ?

MarkRivers, 2018-09-21:
>Is this fixed with ee9fca1 ?

Do you mean the original issue or Andrew's comment above?

tboegi, 2018-09-24:
I'm talking about the original issue.

the comment from Andrew seems to be a different issue to me -
better handled in a different Git commit, and may be in a different issue.

MarkRivers, 2018-09-24:
@tboegi the original issue is fixed.  Soft motors now work on base 7.0.

closed 2018-09-24 by kmpeters