epics-modules/motor #111: The device support init() function is declared incorrectly in many device support files

MarkRivers, 2018-09-09:
EPICS base defines the device support init() function to be the following:

long init(int after);

See section 12.4.2 in https://epics.anl.gov/base/R3-15/5-docs/AppDevGuide/DeviceSupport.html

However, many of the device support files in the motor repository define the init function as:

STATIC long XXX_init(void *);

For example this is the device support init() function for the MAXv in OmsSrc/devMAXV.cc

```
static long MAXv_init(void *arg)
{
    int after = (arg == 0) ? 0 : 1;

    if (*(MAXv_access.init_indicator) == NO)
    {
        errlogSevPrintf(errlogMinor, "%s", errmsg);
        return(ERROR);
    }
    else
        return(motor_init_com(after, MAXv_num_cards, &MAXv_access, &MAXv_cards));
}
```
This is incorrect, because base is calling the init function with an int argument, but this function expects a void* argument.  If an int and a void* are not the same size then this can lead to "after" having the correct value.  This could thus happen on 64-bit systems.  The MAXv is probably currently used only on 32-bit vxWorks systems,  but this is a common error in many device support modules that are run on 64-bit systems.

This is a search of the entire synApps tree for files named dev*.c* which contain the string "init(void".  
```
corvette:~/support>find . -type f -name '*dev*.c*' -exec grep -H "init(void" {} \;
./motor-6-10-1/motorApp/AcsSrc/devMCB4B.cc:STATIC long MCB4B_init(void *);
./motor-6-10-1/motorApp/AcsSrc/devMCB4B.cc:STATIC long MCB4B_init(void *arg)
./motor-6-10-1/motorApp/AcsTech80Src/devSPiiPlus.cc:STATIC long SPiiPlus_init(void *);
./motor-6-10-1/motorApp/AcsTech80Src/devSPiiPlus.cc:STATIC long SPiiPlus_init(void *arg)
./motor-6-10-1/motorApp/DeltaTauSrc/devPmac.cc:static long Pmac_init(void *);
./motor-6-10-1/motorApp/DeltaTauSrc/devPmac.cc:static long Pmac_init(void *arg)
./motor-6-10-1/motorApp/FaulhaberSrc/devMCDC2805.cc:STATIC long MCDC2805_init(void *);
./motor-6-10-1/motorApp/FaulhaberSrc/devMCDC2805.cc:STATIC long MCDC2805_init(void *arg)
./motor-6-10-1/motorApp/ImsSrc/devIM483PL.cc:STATIC long IM483PL_init(void *);
./motor-6-10-1/motorApp/ImsSrc/devIM483PL.cc:STATIC long IM483PL_init(void *arg)
./motor-6-10-1/motorApp/ImsSrc/devIM483SM.cc:STATIC long IM483SM_init(void *);
./motor-6-10-1/motorApp/ImsSrc/devIM483SM.cc:STATIC long IM483SM_init(void *arg)
./motor-6-10-1/motorApp/ImsSrc/devMDrive.cc:STATIC long MDrive_init(void *);
./motor-6-10-1/motorApp/ImsSrc/devMDrive.cc:STATIC long MDrive_init(void *arg)
./motor-6-10-1/motorApp/KohzuSrc/devSC800.cc:static long SC800_init(void *);
./motor-6-10-1/motorApp/KohzuSrc/devSC800.cc:static long SC800_init(void *arg)
./motor-6-10-1/motorApp/MXmotorSrc/devMXmotor.cc:static long MXmotor_init(void *);
./motor-6-10-1/motorApp/MXmotorSrc/devMXmotor.cc:static long MXmotor_init(void *after)
./motor-6-10-1/motorApp/MclennanSrc/devPM304.cc:STATIC long PM304_init(void *);
./motor-6-10-1/motorApp/MclennanSrc/devPM304.cc:STATIC long PM304_init(void *arg)
./motor-6-10-1/motorApp/MicosSrc/devMicos.cc:static long Micos_init(void *);
./motor-6-10-1/motorApp/MicosSrc/devMicos.cc:static long Micos_init(void *arg)
./motor-6-10-1/motorApp/MicroMoSrc/devMVP2001.cc:static long MVP2001_init(void *);
./motor-6-10-1/motorApp/MicroMoSrc/devMVP2001.cc:static long MVP2001_init(void *arg)
./motor-6-10-1/motorApp/NewFocusSrc/devPMNC87xx.cc:STATIC long PMNC87xx_init(void *);
./motor-6-10-1/motorApp/NewFocusSrc/devPMNC87xx.cc:STATIC long PMNC87xx_init(void *arg)
./motor-6-10-1/motorApp/NewportSrc/devMM4000.cc:STATIC long MM4000_init(void *);
./motor-6-10-1/motorApp/NewportSrc/devMM4000.cc:STATIC long MM4000_init(void *arg)
./motor-6-10-1/motorApp/NewportSrc/devPM500.cc:STATIC long PM500_init(void *);
./motor-6-10-1/motorApp/NewportSrc/devPM500.cc:STATIC long PM500_init(void *arg)
./motor-6-10-1/motorApp/NewportSrc/devESP300.cc:static long ESP300_init(void *);
./motor-6-10-1/motorApp/NewportSrc/devESP300.cc:static long ESP300_init(void *arg)
./motor-6-10-1/motorApp/NewportSrc/devMM3000.cc:STATIC long MM3000_init(void *);
./motor-6-10-1/motorApp/NewportSrc/devMM3000.cc:STATIC long MM3000_init(void *arg)
./motor-6-10-1/motorApp/OmsSrc/devOms.cc:static long oms_init(void *);
./motor-6-10-1/motorApp/OmsSrc/devOms.cc:static long oms_init(void *arg)
./motor-6-10-1/motorApp/OmsSrc/devOms58.cc:static long oms_init(void *);
./motor-6-10-1/motorApp/OmsSrc/devOms58.cc:static long oms_init(void *arg)
./motor-6-10-1/motorApp/OmsSrc/devOmsPC68.cc:STATIC long oms_init(void *);
./motor-6-10-1/motorApp/OmsSrc/devOmsPC68.cc:STATIC long oms_init(void *arg)
./motor-6-10-1/motorApp/OmsSrc/devMAXv.cc:static long MAXv_init(void *);
./motor-6-10-1/motorApp/OmsSrc/devMAXv.cc:static long MAXv_init(void *arg)
./motor-6-10-1/motorApp/OrielSrc/devEMC18011.cc:STATIC long EMC18011_init(void *);
./motor-6-10-1/motorApp/OrielSrc/devEMC18011.cc:STATIC long EMC18011_init(void *arg)
./motor-6-10-1/motorApp/PC6KSrc/devPC6K.cc:STATIC long PC6K_init(void *);
./motor-6-10-1/motorApp/PC6KSrc/devPC6K.cc:STATIC long PC6K_init(void *arg)
./motor-6-10-1/motorApp/PiJenaSrc/devPIJEDS.cc:static long PIJEDS_init(void *);
./motor-6-10-1/motorApp/PiJenaSrc/devPIJEDS.cc:static long PIJEDS_init(void *arg)
./motor-6-10-1/motorApp/PiSrc/devPIE516.cc:static long PIE516_init(void *);
./motor-6-10-1/motorApp/PiSrc/devPIE516.cc:static long PIE516_init(void *arg)
./motor-6-10-1/motorApp/PiSrc/devPIC630.cc:static long PIC630_init(void *);
./motor-6-10-1/motorApp/PiSrc/devPIC630.cc:static long PIC630_init(void *arg)
./motor-6-10-1/motorApp/PiSrc/devPIE710.cc:static long PIE710_init(void *);
./motor-6-10-1/motorApp/PiSrc/devPIE710.cc:static long PIE710_init(void *arg)
./motor-6-10-1/motorApp/PiSrc/devPIE816.cc:static long PIE816_init(void *);
./motor-6-10-1/motorApp/PiSrc/devPIE816.cc:static long PIE816_init(void *arg)
./motor-6-10-1/motorApp/PiSrc/devPIC662.cc:static long PIC662_init(void *);
./motor-6-10-1/motorApp/PiSrc/devPIC662.cc:static long PIC662_init(void *arg)
./motor-6-10-1/motorApp/PiSrc/devPIC663.cc:static long PIC663_init(void *);
./motor-6-10-1/motorApp/PiSrc/devPIC663.cc:static long PIC663_init(void *arg)
./motor-6-10-1/motorApp/PiSrc/devPIC844.cc:static long PIC844_init(void *);
./motor-6-10-1/motorApp/PiSrc/devPIC844.cc:static long PIC844_init(void *arg)
./motor-6-10-1/motorApp/PiSrc/devPIC848.cc:static long PIC848_init(void *);
./motor-6-10-1/motorApp/PiSrc/devPIC848.cc:static long PIC848_init(void *arg)
./motor-6-10-1/motorApp/PiSrc/devPIC862.cc:static long PIC862_init(void *);
./motor-6-10-1/motorApp/PiSrc/devPIC862.cc:static long PIC862_init(void *arg)
./motor-6-10-1/motorApp/PiSrc/devPIE517.cc:static long PIE517_init(void *);
./motor-6-10-1/motorApp/PiSrc/devPIE517.cc:static long PIE517_init(void *arg)
./motor-6-10-1/motorApp/SmartMotorSrc/devSmartMotor.cc:static long SmartMotor_init(void *);
./motor-6-10-1/motorApp/SmartMotorSrc/devSmartMotor.cc:static long SmartMotor_init(void *arg)
./motor-6-10-1/motorApp/ThorLabsSrc/devMDT695.cc:STATIC long MDT695_init(void *);
./motor-6-10-1/motorApp/ThorLabsSrc/devMDT695.cc:STATIC long MDT695_init(void *arg)

```
There could be additional files that have the problem but that my search criteria missed.

Note that the problem occurs in many drivers that are used on 64-bit Linux and Windows machines (e.g. MCB4B, many Newport serial and Ethernet controllers, SmartMotor, etc.).

These should all be fixed.

kmpeters, 2018-09-10:
Thanks for creating this issue, @MarkRivers.  I'll fix this before the next release.  It is much easier to resolve this before the motor module split.

kmpeters, 2018-10-09:
> This is incorrect, because base is calling the init function with an int argument, but this function expects a void* argument. If an int and a void* are not the same size then this can lead to "after" having the correct value.

@MarkRivers, I think the app dev guide is wrong.

devMAXv.cc creates this ``devMAXv`` structure, which includes a reference to ``MAXv_init``:
```
struct motor_dset devMAXv =
{
    {8, NULL, MAXv_init, MAXv_init_record, NULL},
    motor_update_values,
    MAXv_start_trans,
    oms_build_trans,
    MAXv_end_trans
};
```

``motor_dset`` is defined in motor.h:
```
/* device support entry table */
struct motor_dset
{
    struct dset base;
    CALLBACK_VALUE (*update_values) (struct motorRecord *);
    long (*start_trans) (struct motorRecord *);
    RTN_STATUS (*build_trans) (motor_cmnd, double *, struct motorRecord *);
    RTN_STATUS (*end_trans) (struct motorRecord *);
};
```

``dset`` is defined in devSup.h (for both base 3.15 and 7.0):
```
typedef struct dset {   /* device support entry table */
    long	number;		/*number of support routines*/
    DEVSUPFUN	report;		/*print report*/
    DEVSUPFUN	init;		/*init support layer*/
    DEVSUPFUN	init_record;	/*init device for particular record*/
    DEVSUPFUN	get_ioint_info;	/* get io interrupt information*/
    /*other functions are record dependent*/
} dset;

```

devSup.h also defines ``DEVSUPFUN``:
```
#ifdef __cplusplus
extern "C" {
    typedef long (*DEVSUPFUN)(void *);	/* ptr to device support function*/
#else
    typedef long (*DEVSUPFUN)();	/* ptr to device support function*/
#endif
```

Unless I'm missing something, this means that the argument to ``MAXv_init`` should be a ``void *``.  This is reinforced by the fact that I changed all the init functions to have int args and I can no longer build any of the model-1 drivers.

kmpeters, 2018-10-09:
Here are the current versions of devSup.h:

https://github.com/epics-base/epics-base/blob/3.15/src/ioc/dbStatic/devSup.h

https://github.com/epics-base/epics-base/blob/7.0/modules/database/src/ioc/dbStatic/devSup.h

MarkRivers, 2018-10-09:
> @MarkRivers, I think the app dev guide is wrong.

No, the app dev guide is correct.  This is the source where the device support init() function is called in iocInit.c in base 7.0.1

```
static void initDevSup(void)
{
    dbRecordType *pdbRecordType;

    for (pdbRecordType = (dbRecordType *)ellFirst(&pdbbase->recordTypeList);
         pdbRecordType;
         pdbRecordType = (dbRecordType *)ellNext(&pdbRecordType->node)) {
        devSup *pdevSup;

        for (pdevSup = (devSup *)ellFirst(&pdbRecordType->devList);
             pdevSup;
             pdevSup = (devSup *)ellNext(&pdevSup->node)) {
            struct dset *pdset = registryDeviceSupportFind(pdevSup->name);

            if (!pdset) {
                errlogPrintf("device support %s not found\n",pdevSup->name);
                continue;
            }
            dbInitDevSup(pdevSup, pdset);   /* Calls pdset->init(0) */
        }
    }
}

static void finishDevSup(void)
{
    dbRecordType *pdbRecordType;

    for (pdbRecordType = (dbRecordType *)ellFirst(&pdbbase->recordTypeList);
         pdbRecordType;
         pdbRecordType = (dbRecordType *)ellNext(&pdbRecordType->node)) {
        devSup *pdevSup;

        for (pdevSup = (devSup *)ellFirst(&pdbRecordType->devList);
             pdevSup;
             pdevSup = (devSup *)ellNext(&pdevSup->node)) {
            struct dset *pdset = pdevSup->pdset;

            if (pdset && pdset->init)
                pdset->init(1);
        }
    }
}
...
void dbInitDevSup(devSup *pdevSup, dset *pdset)
{
    pdevSup->pdset = pdset;
    if (pdevSup->link_type == CONSTANT)
        pdevSup->pdsxt = &devSoft_DSXT;

    if (pdset->init) {
        pthisDevSup = pdevSup;
        pdset->init(0);
        pthisDevSup = NULL;
    }
}
```

Note that pdset->init() is being called with the integer values 0 and 1, not a pointer.

You can see this even more clearly with get_ioint_info.  By your argument it should be called with a single void* argument.  But this is the declaration of that function in asyn/devAsynInt32.c
'''
static long getIoIntInfo(int cmd, dbCommon *pr, IOSCANPVT *iopvt)
'''

anjohnson, 2018-10-09:
@kmpeters `struct dset` was written in the days of K&R C which didn't need you to specify arguments for pointers to functions (or even for regular functions), so `DEVSUPFUN` was originally defined as a generic function pointer without specifying what the arguments would be. When C++ came along it didn't support that old K&R syntax, so someone added the C++-specific typedef that you see, but the argument it uses is most definitely wrong.

Unfortunately fixing devSup.h is not a trivial thing to do given the amount of code out there that includes it. We have a [branch in review](https://code.launchpad.net/~epics-core/epics-base/+git/dtypedset/+merge/332730) that defines the function pointers properly and is being designed to allow modules to be modified incrementally so we are finally working on this, but it hasn't been merged just yet.

kmpeters, 2018-10-10:
So that means we can't solve this problem until devSup.h is fixed:
> If an int and a void* are not the same size then this can lead to "after" having the correct value. This could thus happen on 64-bit systems.

But changing the argument to the XXX_init functions to an int still has value because it is technically correct, even though it doesn't solve the problem yet; it is a change that will be required when devSup.h is fixed.

kmpeters, 2018-10-10:
I'm not sure what I mean by that and I don't understand the build problems I'm seeing.  I'm very confused.

I've changed all of the init functions to accept int arguments:
https://github.com/kmpeters/motor/tree/issue111

When I try to build my issue111 branch, some of the drivers (DeltaTauSrc,OmsSrc) fail to build:
```
../devPmac.cc:66: error: invalid conversion from 'long int (*)(int)' to 'long int (*)(void*)'
make[3]: *** [devPmac.o] Error 1
make[3]: Leaving directory `/net/s100dserv/xorApps/epics/motor-devel/support/motor-kmp/motorApp/DeltaTauSrc/O.vxWorks-ppc32'
make[2]: *** [install.vxWorks-ppc32] Error 2
make[2]: Leaving directory `/net/s100dserv/xorApps/epics/motor-devel/support/motor-kmp/motorApp/DeltaTauSrc'
make[1]: *** [DeltaTauSrc.install] Error 2
make[1]: Leaving directory `/net/s100dserv/xorApps/epics/motor-devel/support/motor-kmp/motorApp'
make: *** [motorApp.install] Error 2
```
```
../devOmsPC68.cc:74:1: error: invalid conversion from ‘long int (*)(int)’ to ‘DEVSUPFUN {aka long int (*)(void*)}’ [-fpermissive]
 };
 ^
make[1]: *** [devOmsPC68.o] Error 1
make[1]: Leaving directory `/net/s100dserv/xorApps/epics/motor-devel/support/motor-kmp/motorApp/OmsSrc/O.linux-x86_64'
make: *** [install.linux-x86_64] Error 2
```
While others (SoftMotorSrc, ThorLabsSrc, NewportSrc) succeed.

kmpeters, 2018-10-10:
The compile errors go away if I explicitly cast the init function as (DEVSUPFUN) in the motor_dset structure:
```
struct motor_dset devOmsPC68 =
{
    {8, NULL, (DEVSUPFUN) oms_init, oms_init_record, NULL},
    motor_update_values,
    oms_start_trans,
    oms_build_trans,
    oms_end_trans
};
```

kmpeters, 2018-10-10:
If the dset structure defines a function as having a void * argument, but an integer is passed to the device support and the init function accepts an integer argument, is there still the possibility of a problem due to differing sizes of void * and int on 64-bit systems?  Do we need the devSup.h fix from the base branch that is in review for the problem to be fully resolved?

kmpeters, 2018-10-10:
Continuing my thinking out loud: if the function pointer in the dset structure is the same size regardless of the type or arguments, then the changing the init argument to int and casting the init function as (DEVSUPFUN) does avoid problems due to void * and int having different sizes on 64-bit systems.  We don't need the devSup.h fix from the base branch, but it would be nice to keep things consistent.

PR #116 (2018-10-16): Changed init function arguments to int
- Changed init function arguments in model-1 drivers to int
- Simplified the init function logic related to "after"
- Cast the pointer to the init function as  DEVSUPFUN in the motor_dset structure

Fixes #111 
Fixes #115
closed 2018-10-16 by kmpeters