epics-modules/motor #113: Setpoint Deadband Field SPDB

mp49, 2018-09-19:
Hi,

I discussed this a while back with @tboegi but I don't think it ever made it into the motor record. 

A number of our users and staff have requested that the motor record have a setpoint deadband field (eg. SPDB), which prevents any motion from happening if the readback position is within SPDB. Without a SPDB we have issues with motors moving unnecessarily. For example, if the RBV is 0.001 and we set VAL to 0.0, if we have a non-zero backlash correction then the motor will move unnecessarily to take out backlash. Even without backlash correction we often enable and disable the drive amplifier for no good reason.

Currently the effective setpoint deadband is equal to MRES, but this is often too small to be an effective setpoint deadband.

The new SPDB field would default to 0.0

I've implemented the change locally at the SNS and it's been in use for several weeks on several beamlines. If this change sounds ok then I will make a pull request. I've pasted the code below for preliminary review. Ignore the line numbers below, but it's in the section in do_work() that decides if a move is too small or not.

```
diff --git a/motorApp/MotorSrc/motorRecord.cc b/motorApp/MotorSrc/motorRecord.cc
index c7dae83..a238b93 100644
--- a/motorApp/MotorSrc/motorRecord.cc
+++ b/motorApp/MotorSrc/motorRecord.cc
@@ -2137,6 +2137,7 @@ static RTN_STATUS do_work(motorRecord * pmr, CALLBACK_VALUE proc_ind)
             double vbase = pmr->vbas / fabs(pmr->mres); /* base speed      */
             double vel = pmr->velo / fabs(pmr->mres);   /* normal speed    */
             double acc = (vel - vbase) / pmr->accl;     /* normal accel.   */
+            double spdbval = pmr->spdb / pmr->mres;     /* setpoint deadband */
             /*
              * 'bpos' is one backlash distance away from 'newpos'.
              */
@@ -2148,7 +2149,7 @@ static RTN_STATUS do_work(motorRecord * pmr, CALLBACK_VALUE proc_ind)
             double relbpos = ((pmr->dval - pmr->bdst) - pmr->drbv) / pmr->mres;
             double rbdst1 = 1.0 + (fabs(pmr->bdst) / fabs(pmr->mres));
             long rdbdpos = NINT(pmr->rdbd / fabs(pmr->mres)); /* retry deadband steps */
-            long rpos, npos;
+            long rpos, npos, spdb;
             msta_field msta;
             msta.All = pmr->msta;
 

@@ -2176,6 +2177,18 @@ static RTN_STATUS do_work(motorRecord * pmr, CALLBACK_VALUE proc_ind)
             rpos = NINT(rbvpos);
             npos = NINT(newpos);
             too_small = false;
+
+            /* Don't move if new setpoint is within SPDB of DRBV */
+            /* MRP 8/9/18 */
+            spdb = NINT(spdbval);
+            if ((pmr->mip & MIP_RETRY) == 0) {
+              if (spdb > 0) {
+                if (((npos - spdb) < rpos) && ((npos + spdb) > rpos)) {
+                  too_small = true;
+                }
+              }
+            }
+
             if ((pmr->mip & MIP_RETRY) == 0)
```

and in the DBD file:

```
+        field(SPDB,DBF_DOUBLE) {
+                prompt("Setpoint Deadband (EGU)")
+                promptgroup(GUI_COMMON)
+                special(SPC_MOD)
+                interest(1)
+        }
```

tboegi, 2018-09-20:
The change looks good - as far as I can see at first glance.
Do you have your code on a public repo ?
Otherwise I am looking forward to review a PR, as we have an SDBD field in the
ESS version of motorRecord.

kmpeters, 2018-09-20:
Why is the existing retry deadband (RDBD) field insufficient?

SDBD would govern new moves and RDBD would govern retries.  Does it make sense to ever allow RDBD to be greater than SDBD?

mp49, 2018-09-20:
I had the same thought about RDBD, but I'm not sure it will cover all cases, so adding a separate tolerance field give us most flexibility.

The way I implemented it SPDB does cover new moves, but RDBD still covers retries. So you could have a larger SPDB, but it will still try to position within RDBD if it does move.

I'm not sure RDBD>SPDB makes sense, but I could imagine having SPDB>=RDBD (and that's what people have asked for).

kmpeters, 2018-09-20:
> I'm not sure RDBD>SPDB makes sense, but I could imagine having SPDB>=RDBD (and that's what people have asked for).

If SPDB > RDBD, and a poorly-behaved motor gets within the SPDB of the VAL, but not within the RDBD, a user won't be able to command it to go to that VAL without first sending it to a different position.

kmpeters, 2018-09-20:
Didn't the original argument for the SPDB field involve motors with encoders or external readback PVs?

mp49, 2018-09-20:
I think both those statement are true. If a motor wanders to within SPDB (of the current DRBV readback) then they can't move it. I think it only applies to motors+encoders.

tboegi, 2018-09-21:
The use of abs() gives always a compiler warning.
Trying to be helpful, I added a slightly different version.

https://github.com/epics-modules/motor/pull/114
@mp49, @kmpeters please let me know what you think

kmpeters, 2018-12-11:
Merged in commit 6f2f42305a30e92ab81a34a9bc583bf0f543cb3e

closed 2018-12-11 by kmpeters