epics-modules/motor #117: Travis builds against base-3.15 are broken

kmpeters, 2018-10-10:
The problem appears to be due to base, not motor:
```
/usr/bin/gcc  -D_GNU_SOURCE -D_DEFAULT_SOURCE            -D_X86_64_  -DUNIX  -Dlinux     -O3 -g   -Wall      -mtune=generic     -m64 -fPIC -I. -I../O.Common -I. -I../../../src/libCom/osi/compiler/gcc -I../../../src/libCom/osi/compiler/default -I. -I../../../src/libCom/osi/os/Linux -I../../../src/libCom/osi/os/posix -I../../../src/libCom/osi/os/default -I.. -I../../../src/libCom/as -I../../../src/libCom/bucketLib -I../../../src/libCom/calc -I../../../src/libCom/cvtFast -I../../../src/libCom/cppStd -I../../../src/libCom/cxxTemplates -I../../../src/libCom/dbmf -I../../../src/libCom/ellLib -I../../../src/libCom/env -I../../../src/libCom/error -I../../../src/libCom/fdmgr -I../../../src/libCom/flex -I../../../src/libCom/freeList -I../../../src/libCom/gpHash -I../../../src/libCom/iocsh -I../../../src/libCom/log -I../../../src/libCom/macLib -I../../../src/libCom/misc -I../../../src/libCom/osi -I../../../src/libCom/pool -I../../../src/libCom/ring -I../../../src/libCom/taskwd -I../../../src/libCom/timer -I../../../src/libCom/yacc -I../../../src/libCom/yacc -I../../../src/libCom/yajl -I../../../include/compiler/gcc -I../../../include/os/Linux -I../../../include         -MM -MF asLib.d  asLib.c
make[3]: Leaving directory `/home/travis/.cache/support/epics-base/src/libCom/O.linux-x86_64'
make[3]: Entering directory `/home/travis/.cache/support/epics-base/src/libCom/O.linux-x86_64'
make[3]: *** No rule to make target `install'.  Stop.
make[3]: Leaving directory `/home/travis/.cache/support/epics-base/src/libCom/O.linux-x86_64'
make[2]: *** [install.linux-x86_64] Error 2
make[2]: Leaving directory `/home/travis/.cache/support/epics-base/src/libCom'
make[1]: *** [libCom.install] Error 2
make[1]: Leaving directory `/home/travis/.cache/support/epics-base/src'
make: *** [src.install] Error 2
make: Leaving directory `/home/travis/.cache/support/epics-base'
The command "./.ci/travis-prepare.sh" failed and exited with 2 during .
Your build has been stopped.
```
But the travis script may require modification.

kmpeters, 2018-10-10:
@keenanlang, should we be using a released version of base rather than the master branch?

anjohnson, 2018-10-10:
It's not too often that the Base builds are broken, but you might want to switch to a tagged release instead to avoid problems when they are. The Base-3.15 branch is now fixed.

kmpeters, 2018-10-10:
Fixed by 8c412e7db450c2df8392d959f61e24b72eef2334 and 66203a98cb2701f5be68f0f2eaffef74b3bb3131

closed 2018-10-10 by kmpeters