epics-modules/motor #121: Installation and building instructions

nilsmelchert, 2018-11-19:
Hello,
I am trying to work with the smc100 Newport controller but have a hard time building this package. Is there a detailed installation/building instruction somewhere?
Am I able to use only the smc100 source code?

Thanks a lot for your replys

MarkRivers, 2018-11-19:
Can you provide more detail on what problems you are having?  What architecture are you building for, etc.  It builds fine for me on linux-x86_64, with just one warning:
```
/usr/bin/g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE         -Wno-write-strings   -D_X86_64_  -DUNIX  -Dlinux     -O3 -g   -Wall      -mtune=generic     -m64 -fPIC -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/gcc -I../../../include/os/Linux -I../../../include      -I/corvette/home/epics/devel/asyn-4-33/include   -I/corvette/home/epics/devel/seq-2-2-5/include   -I/corvette/home/epics/devel/busy-1-7/include -I/corvette/usr/local/epics-devel/base-7.0.1/include/compiler/gcc -I/corvette/usr/local/epics-devel/base-7.0.1/include/os/Linux -I/corvette/usr/local/epics-devel/base-7.0.1/include   -I/corvette/home/epics/devel/ipac-2-14/include        -c ../SMC100Driver.cpp
../SMC100Driver.cpp: In constructor 'SMC100Controller::SMC100Controller(const char*, const char*, int, double, double, double)':
../SMC100Driver.cpp:52:15: warning: variable 'pAxis' set but not used [-Wunused-but-set-variable]
   SMC100Axis *pAxis;
               ^
```
Can you send the output when you try to build it?

kmpeters, 2018-12-11:
@nilsmelchert, are you still having trouble building support for the SMC100?

closed 2020-06-18 by kmpeters