epics-modules/motor #123: soft motor default LLM same as HLM

prjemian, 2018-12-04:
With motor-R6-10-1 and using the motorSim soft motors, both LLM and HLM come up as identical. Since they are non-zero and the motor is at zero, the soft motor cannot move until the lower limit is changed. All 16 soft motors created from an XXX IOC have this problem.  (See https://github.com/epics-modules/xxx/issues/18)

The problem seems to be when a symbol is not defined and a default value is provided.  In the source of the [`motorSim.iocsh`](https://github.com/epics-modules/motor/blob/master/iocsh/motorSim.iocsh#L29):

    motorSimCreate($(CONTROLLER=0), 0, $(LOW_LIM=-32000), $(HIGH_LIM=32000), $(HOME_POS=0), 1, $(NUM_AXES=1))

The default value for `LOW_LIM`, when the symbol is not defined, becomes positive.

kmpeters, 2018-12-11:
@prjemian, does commit df11f3e153e2aa6c6cb605cbc1392e0a9e663d93 resolve this issue and eliminate the need to define LOW_LIM in the iocshLoad call in motors.cmd?

prjemian, 2018-12-11:
@kmpeters - I confirm that is the fix.  Thanks!

closed 2018-12-11 by kmpeters