epics-modules/motor #133: LGTM.com code review

prjemian, 2019-05-23:
[LGTM.com](https://lgtm.com/projects/g/epics-modules/motor/) is ready to do code review on pull requests for [epics-modules/motor](https://github.com/epics-modules/motor).  Initiate the first one with a trivial edit.

prjemian, 2019-05-23:
Seems that we are probably out of luck using LGTM.com for automated code reviews.  The [first build](https://lgtm.com/projects/g/epics-modules/motor/logs/languages/lang:cpp) for epics-modules/motor has failed due to this error:

    "Semmle autobuild: no supported build system detected."

means it cannot deal with the EPICS build system.

Too bad, it was a neat idea.

closed 2019-05-23 by prjemian

anjohnson, 2019-05-23:
Hi Pete, It looks like you need to configure it with an `lgtm.yml` or `.lgtm.yml` file, similar to the `.travis.yml` files that you already have. See [this help document](https://lgtm.com/help/lgtm/lgtm.yml-configuration-file) for details...