epics-modules/motor #145: Unable to Build Motor Modules

DanielALS, 2019-09-11:
Building fresh synApps 6.1 using the APS install script.
When make gets to Motor, getting the error below. I think a macro is not getting picked up or I failed to set one. Being a generic installation, I'm trying to keep all motor types for the time being. 

Linux CentOS 7 64 bit.

Thank you

```
make[1]: Entering directory `/usr/local/epics/R3.15.4/modules/soft/synApps_6_1/support/motor-R7-1/modules'
make -C ./motorOms install
make[2]: Entering directory `/usr/local/epics/R3.15.4/modules/soft/synApps_6_1/support/motor-R7-1/modules/motorOms'
configure/CONFIG:17: /configure/CONFIG: No such file or directory
configure/RULES_DIRS:2: /configure/RULES_DIRS: No such file or directory
make[2]: *** No rule to make target `/configure/RULES_DIRS'.  Stop.
make[2]: Leaving directory `/usr/local/epics/R3.15.4/modules/soft/synApps_6_1/support/motor-R7-1/modules/motorOms'
make[1]: *** [motorOms.install] Error 2
make[1]: Leaving directory `/usr/local/epics/R3.15.4/modules/soft/synApps_6_1/support/motor-R7-1/modules'
make: *** [modules.install] Error 2
```

kmpeters, 2019-09-13:
I have no problems building synApps_6_1 against base-3.15.4 using the following procedure (using tcsh):
```shell
$ mkdir github-motor-issue145
$ cd github-motor-issue145
$ wget https://epics.anl.gov/download/base/base-3.15.4.tar.gz
$ tar xzvf base-3.15.4.tar.gz
$ cd base-3.15.4
$ setenv EPICS_HOST_ARCH linux-x86_64
$ make
$ cd ..
$ wget https://raw.githubusercontent.com/EPICS-synApps/support/master/assemble_synApps.sh
$ vim assemble_synApps.sh
(manully replace path to EPICS_BASE with absolute path to base-3.15.4)
$ chmod a+x assemble_synApps.sh
$ ./assemble_synApps.sh
$ cd synApps/support
$ make
```
If I look at changes to the motor module, the only changes are see are to the RELEASE file, which are the result of the 'make release' call in assemble_synApps.sh:
```shell
$ cd motor-R7-1/
$ git status
# Not currently on any branch.
# Changes not staged for commit:
#   (use "git add <file>..." to update what will be committed)
#   (use "git checkout -- <file>..." to discard changes in working directory)
#
#	modified:   configure/RELEASE
#
no changes added to commit (use "git add" and/or "git commit -a")
$ git diff
diff --git a/configure/RELEASE b/configure/RELEASE
index 510fd57..8b0f6bf 100644
--- a/configure/RELEASE
+++ b/configure/RELEASE
@@ -9,22 +9,22 @@
 #
 # Add your declarations here
 
-SUPPORT=
+SUPPORT=/home/kmpeters/github-motor-issue145/synApps/support
 -include $(TOP)/../configure/SUPPORT.$(EPICS_HOST_ARCH)
 
 # If any motor controller communication mechanism besides 
 # the VME backplane is required, then ASYN must be defined.
 # Recommended ASYN release: R4-35
-ASYN=$(SUPPORT)/asyn-R4-35
+ASYN=$(SUPPORT)/asyn-R4-36
 
 # Need the sequencer and the busy record for trajectory scanning
 # Recommended SNCSEQ release: R2.2.6
 SNCSEQ=$(SUPPORT)/seq-2-2-6
 # Recommended BUSY release: R1-7-1
-BUSY=$(SUPPORT)/busy-R1-7-1
+BUSY=$(SUPPORT)/busy-R1-7-2
 
 # Recommended EPICS release: R3.15.6
-EPICS_BASE=
+EPICS_BASE=/home/kmpeters/github-motor-issue145/base-3.15.4
 -include $(TOP)/../configure/EPICS_BASE.$(EPICS_HOST_ARCH)
 
 # The following must be defined for motorMXmotor
@@ -32,7 +32,7 @@ EPICS_BASE=
 
 # IPAC is required to build motorHytec
 # Recommended IPAC release: R2-15
-IPAC=$(SUPPORT)/ipac-R2-15
+IPAC=$(SUPPORT)/ipac-2-15
 
 # Script module needed to build motorScriptMotor
 #!LUA=$(SUPPORT)/lua-R1-3
```

kmpeters, 2019-09-13:
What does this file contain?
```
/usr/local/epics/R3.15.4/modules/soft/synApps_6_1/support/motor-R7-1/modules/RELEASE.linux-x86_64.local
```
If I clear my path to EPICS base in RELEASE.linux-x86_64.local, I see the error in your original post:
```
$ cat ../RELEASE.linux-x86_64.local
MOTOR = /home/kmpeters/github-motor-issue145/synApps/support/motor-R7-1
ASYN = /home/kmpeters/github-motor-issue145/synApps/support/asyn-R4-36
SNCSEQ = /home/kmpeters/github-motor-issue145/synApps/support/seq-2-2-6
BUSY = /home/kmpeters/github-motor-issue145/synApps/support/busy-R1-7-2
IPAC = /home/kmpeters/github-motor-issue145/synApps/support/ipac-2-15
EPICS_BASE = 
$ make
<snip>
make[2]: Entering directory `/home/kmpeters/github-motor-issue145/synApps/support/motor-R7-1/modules/motorOms'
configure/CONFIG:17: /configure/CONFIG: No such file or directory
configure/RULES_DIRS:2: /configure/RULES_DIRS: No such file or directory
make[2]: *** No rule to make target `/configure/RULES_DIRS'.  Stop.
make[2]: Leaving directory `/home/kmpeters/github-motor-issue145/synApps/support/motor-R7-1/modules/motorOms'
make[1]: *** [motorOms.install] Error 2
make[1]: Leaving directory `/home/kmpeters/github-motor-issue145/synApps/support/motor-R7-1/modules'
make: *** [modules.install] Error 2
```

DanielALS, 2019-09-14:
Thank you for going through all that. I'll get back to this Monday morning. Cheers.

DanielALS, 2019-09-16:
Problem solved, the `modules/RELEASE.linux-x86_64.local` had a path that didn't resolve. Instead of failing, Make just blanked it out. I really don't understand how the path got messed up. 

Cheers

kmpeters, 2019-09-16:
I'm glad to hear the issue is resolved.

closed 2019-09-16 by kmpeters