epics-modules/motor #149: motor/configure/RELEASE problem on Windows

MarkRivers, 2019-10-28:
motor/configure/RELEASE does not work correctly building Windows in same tree as Linux

I believe these lines are intended to support that:
```
# These lines allow developers to override these RELEASE settings
# without having to modify this file directly.
-include $(or $(MOTOR),$(TOP))/../RELEASE.local
-include $(or $(MOTOR),$(TOP))/../RELEASE.$(EPICS_HOST_ARCH).local
-include $(or $(MOTOR),$(TOP))/configure/RELEASE.local
```
However, they don't work.  I have a file `synApps/support/RELEASE.windows-x64-static.local` that defines SUPPORT and EPICS_BASE using Windows syntax.  That file is clearly not being read when I build for Windows in modules/motorOms, for example.

It does work if I replace the lines above with the standard lines used in all other modules:
```
-include $(TOP)/../RELEASE.local
-include $(TOP)/../RELEASE.$(EPICS_HOST_ARCH).local
-include $(TOP)/configure/RELEASE.local
```
What is the  $(or $(MOTOR),$(TOP)) syntax trying to accomplish, and why isn't it working?

kmpeters, 2019-10-30:
They fix a recursive loop when building motor drivers outside of the motor module: https://github.com/epics-modules/motor/commit/ef56405411df6d032a7a9d313a9cab8f5425e0f1

MarkRivers, 2019-10-30:
Sorry, that file did exist and contains this:
```
corvette:~/devel/motor/modules>more RELEASE.windows-x64-static.local
MOTOR = J:/epics/devel/motor
ASYN = J:/epics/devel/asyn-4-37
SNCSEQ = J:/epics/devel/seq-2-2-5
BUSY = J:/epics/devel/busy-1-7-2
IPAC = J:/epics/devel/ipac-2-15
EPICS_BASE = H:/epics-devel/base-7.0.3
```

Here is the error if I use the master branch of motor.  I did
```
make -sj realclean realuninstall
make
```
on windows-x64-static.  I get this error:
```
make[4]: Entering directory 'J:/epics/devel/motor/modules/motorOms/configure/O.windows-x64-static'
perl -CSD H:/epics-devel/base-7.0.3/bin/windows-x64-static/convertRelease.pl checkRelease

Definition of IPAC conflicts with MOTOR support.
In this application or module, a RELEASE file
conflicts with MOTOR at J:/epics/devel/motor
  Here: IPAC = J:/epics/devel/ipac-2-15
  MOTOR: IPAC = /corvette/home/epics/devel/ipac-2-15
Definition of EPICS_BASE conflicts with MOTOR support.
In this application or module, a RELEASE file
conflicts with MOTOR at J:/epics/devel/motor
  Here: EPICS_BASE = H:/epics-devel/base-7.0.3
  MOTOR: EPICS_BASE = /corvette/usr/local/epics-devel/base-7.0.3
Definition of BUSY conflicts with MOTOR support.
In this application or module, a RELEASE file
conflicts with MOTOR at J:/epics/devel/motor
  Here: BUSY = J:/epics/devel/busy-1-7-2
  MOTOR: BUSY = /corvette/home/epics/devel/busy-1-7-2
Definition of ASYN conflicts with MOTOR support.
In this application or module, a RELEASE file
conflicts with MOTOR at J:/epics/devel/motor
  Here: ASYN = J:/epics/devel/asyn-4-37
  MOTOR: ASYN = /corvette/home/epics/devel/asyn-4-37
Definition of SNCSEQ conflicts with MOTOR support.
In this application or module, a RELEASE file
conflicts with MOTOR at J:/epics/devel/motor
  Here: SNCSEQ = J:/epics/devel/seq-2-2-5
  MOTOR: SNCSEQ = /corvette/home/epics/devel/seq-2-2-5

make[4]: *** [H:/epics-devel/base-7.0.3/configure/RULES_BUILD:191: checkRelease] Error 1
make[4]: Leaving directory 'J:/epics/devel/motor/modules/motorOms/configure/O.windows-x64-static'
make[3]: *** [H:/epics-devel/base-7.0.3/configure/RULES_ARCHS:58: install.windows-x64-static] Error 2
make[3]: Leaving directory 'J:/epics/devel/motor/modules/motorOms/configure'
make[2]: *** [H:/epics-devel/base-7.0.3/configure/RULES_DIRS:84: configure.install] Error 2
make[2]: Leaving directory 'J:/epics/devel/motor/modules/motorOms'
make[1]: *** [H:/epics-devel/base-7.0.3/configure/RULES_DIRS:84: motorOms.install] Error 2
make[1]: Leaving directory 'J:/epics/devel/motor/modules'
make: *** [H:/epics-devel/base-7.0.3/configure/RULES_DIRS:84: modules.install] Error 2
```

However, if I change the motor/configure/RELEASE to use the standard syntax for the top-level includes it works fine.

MarkRivers, 2019-10-30:
The error when I use master indicates that when check_release runs in motor/ it is not correctly including $(TOP)/../RELEASE.windows-x64-static.local.

When I use the modified version of motor/configure/RELEASE it does include that file correctly.

kmpeters, 2019-10-30:
That is a checkRelease error that can be avoided by setting ``CHECK_RELEASE = WARN`` in motor's ``CONFIG_SITE.local``

MarkRivers, 2019-10-30:
>The check release script isn't smart enough to handle the statements at the end of motor's RELEASE file.

That is not true. If I replace those lines with the standard ones then check release correctly reads the file in $(TOP)/../RELEASE.windows-x64-static.local.  If it did not I would receive a check release error with that version of RELEASE and I do not.

I don’t want to disable check release, that is a kludge that invites errors later.

Please explain what that (or ...) syntax in RELEASE  is supposed to do. It does not seem to be working correctly.

kmpeters, 2019-10-30:
> Please explain what that (or ...) syntax in RELEASE is supposed to do. It does not seem to be working correctly.

If you build motor without any of the driver submodules, using a ``RELEASE`` file with these statements:
```
-include $(TOP)/../RELEASE.local
-include $(TOP)/../RELEASE.$(EPICS_HOST_ARCH).local
-include $(TOP)/configure/RELEASE.local
```
And then you try to build a driver module outside of the motor tree, using the driver modules's ``EXAMPLE_RELEASE.local``, the driver module will fail to build.  There will be many warnings about recursive loops:
```
make -C O.linux-x86_64 -f ../Makefile TOP=../.. \
    T_A=linux-x86_64 install
Release.pm: Recursive loop found in RELEASE files,
discovered in ../../configure/RELEASE.local
make[2]: Entering directory `/net/s100dserv/xorApps/epics/motor-devel/issue149/motorOms/configure/O.linux-x86_64'
perl -CSD /APSshare/epics/base-3.15.6/bin/linux-x86_64/convertRelease.pl checkRelease
Release.pm: Recursive loop found in RELEASE files,
discovered in /net/s100dserv/xorApps/epics/motor-devel/issue149/motorOms/configure/RELEASE.local
make[2]: [warnRelease] Error 2 (ignored)
make[2]: Leaving directory `/net/s100dserv/xorApps/epics/motor-devel/issue149/motorOms/configure/O.linux-x86_64'
```
And the build will ultimately fail with an error like this:
```
make[3]: *** No rule to make target `../../../lib/linux-x86_64/libmotor.a', needed by `liboms.a'.  Stop.
make[3]: Leaving directory `/net/s100dserv/xorApps/epics/motor-devel/issue149/motorOms/omsApp/src/O.linux-x86_64'
make[2]: *** [install.linux-x86_64] Error 2
make[2]: Leaving directory `/net/s100dserv/xorApps/epics/motor-devel/issue149/motorOms/omsApp/src'
make[1]: *** [src.install] Error 2
make[1]: Leaving directory `/net/s100dserv/xorApps/epics/motor-devel/issue149/motorOms/omsApp'
make: *** [omsApp.install] Error 2
```
Because:
1. ``motorOms/configure/EXAMPLE_RELEASE.local`` includes ``$(MOTOR)/configure/RELEASE``
2. ``motor/configure/RELEASE"`` includes ``$(TOP)/configure/RELEASE.local``
3. But ``TOP`` is ``motorOms`` in this case, NOT ``motor``

MarkRivers, 2019-10-30:
If I am building in the distributed configuration with all modules in the tree then should motor/configure/RELEASE be including synApps/support/RELEASE.$(EPICS_HOST_ARCH).local? It does not do so when check release is run building the submodules, but it does if I use the standard syntax.

It seems that work is needed to make it work in both situations.

kmpeters, 2019-10-30:
I think I can resolve this issue without breaking builds of external drivers by replacing this line in every driver's ``EXAMPLE_RELEASE.local``:
```
$ git diff
diff --git a/configure/EXAMPLE_RELEASE.local b/configure/EXAMPLE_RELEASE.local
index 4012509..2709d1a 100644
--- a/configure/EXAMPLE_RELEASE.local
+++ b/configure/EXAMPLE_RELEASE.local
@@ -1,4 +1,4 @@
 MOTOR=/home/oxygen/KPETERSN/epics/motor-split/motor-kmp
--include $(MOTOR)/configure/RELEASE
+-include $(MOTOR)/modules/RELEASE.$(EPICS_HOST_ARCH).local
 # path to motorOms is needed to build the IOC inside motorOms, but outside motor
 MOTOR_OMS=/home/oxygen/KPETERSN/epics/motor-split/OMS/motorOms4
```
And then reverting the change in commit https://github.com/epics-modules/motor/commit/ef56405411df6d032a7a9d313a9cab8f5425e0f1

PR #150 (2019-11-04): Fix for issue 149
Reverted the commit that caused issue 149 and updated the driver modules' EXAMPLE_RELEASE.local files to avoid problems when building them outside the motor tree.

Fixes #149
commit fe8f8159 (2019-11-04): Merge pull request #150 from epics-modules/issue149

Reverted the commit that caused issue 149 and updated the driver modules' EXAMPLE_RELEASE.local files to avoid problems when building them outside the motor tree.

Fixes #149
closed 2019-11-04 by kmpeters