epics-modules/motor #166: MOTOR_REC_{OFFSET,RESOLUTION,DIRECTION} errors at iocInit for asyn model-2 drivers

kmpeters, 2020-07-31:
If ``asyn_motor.db`` or ``basic_asyn_motor.db`` is used with an asyn model-2 driver, the follow errors occur at iocInit:
```
kmp2:m1Offset devAsynFloat64::initCommon drvUserCreate drvmotorAsyn::drvUserCreate, unknown command=MOTOR_REC_OFFSET
Bad init_rec return value  PV: kmp2:m1Offset ao: init_record

kmp2:m1Resolution devAsynFloat64::initCommon drvUserCreate drvmotorAsyn::drvUserCreate, unknown command=MOTOR_REC_RESOLUTION
Bad init_rec return value  PV: kmp2:m1Resolution ao: init_record

kmp2:m1Direction devAsynInt32::initCommon drvUserCreate drvmotorAsyn::drvUserCreate, unknown command=MOTOR_REC_DIRECTION
```
The records that cause the errors were added in this commit: https://github.com/epics-modules/motor/commit/d091a0917525e9ff899946234506254933098b3a
After this change to the asynMotor parameters: https://github.com/epics-modules/motor/commit/102f74b5842f8cdea9fff5f74264c36f633411e3

One way to resolve this issue would be to add the new parameters to ``motorAxisParam_t`` in ``motor_interface.h`` so they can be added to ``motorCommand`` enum and  ``motorCommandStruct`` in ``drvMotorAsyn.c``.

A simpler way to resolve this issues is to create ``asyn_2_motor.db`` and ``basic_asyn_2_motor.db`` that omit the records added in https://github.com/epics-modules/motor/commit/d091a0917525e9ff899946234506254933098b3a and use them with asyn model-2 drivers, like motorMotorSim.

kmpeters, 2020-07-31:
@MarkRivers, do you have thoughts on this issue?

MarkRivers, 2020-08-01:
What other Model 2 drivers are in use?  There is a Model 3 version of the simulation motor, correct?

MarkRivers, 2020-08-02:
I searched all of motor and motor/modules for motor_interface.h.  
```
corvette:~/devel/motor>find . -name '*.c*' -exec grep -H motor_interface.h {} \;
./modules/motorAerotech/aerotechApp/src/drvA3200Asyn.cc:#include "motor_interface.h"
./modules/motorAerotech/aerotechApp/src/drvEnsembleAsyn.cc:#include "motor_interface.h"
./modules/motorMotorSim/motorSimApp/src/devMotorSim.c:#include "motor_interface.h"
./modules/motorMotorSim/motorSimApp/src/drvMotorSim.c:#include "motor_interface.h"
./modules/motorPIGCS2/pigcs2App/src/PIasynAxis.cpp:#include <motor_interface.h>
./modules/motorPIGCS2/pigcs2App/src/PIasynController.cpp:#include <motor_interface.h>
./modules/motorAttocube/attocubeApp/src/drvANC150Asyn.cc:#include "motor_interface.h"
./modules/motorNewport/newportApp/src/drvMM4000Asyn.c:#include "motor_interface.h"
./modules/motorNewport/newportApp/src/drvXPSAsyn.c:#include "motor_interface.h"
./motorApp/MotorSrc/devMotorAsyn.c:#include "motor_interface.h"
./motorApp/MotorSrc/drvMotorAsyn.c:#include "motor_interface.h"
./motorApp/MotorSrc/drvMotorAsyn.c:    /* Parameters - these must match the definitions in motor_interface.h */

corvette:~/devel/motor>find . -name '*.h' -exec grep -H motor_interface.h {} \;
./modules/motorNewport/newportApp/src/XPSAsynInterpose.h:#include "motor_interface.h"
```

Most of these appear to be Model 2 drivers, although motorPIGCS2 appears to be a Model 3 driver and should not need motor_interface.h.

tboegi, 2020-08-12:
Please allow a side-question, how do I read this 2 contradicting lines ?
...Model 3 driver and should not need motor_interface.h
./motorApp/MotorSrc/drvMotorAsyn.c:#include "motor_interface.h"

I tried to remove all code in drvMotorAsyn.c, but then I ran into this error when compiling/linking
an IOC with ethercatmc:

Undefined symbols for architecture x86_64:
  "_pvar_func_motorRegister", referenced from:

MarkRivers, 2020-08-12:
>Please allow a side-question, how do I read this 2 contradicting lines ?
...Model 3 driver and should not need motor_interface.h
./motorApp/MotorSrc/drvMotorAsyn.c:#include "motor_interface.h"

I don't think the lines are contradictory.  drvMotorAsyn.c is not used by Model 3 drivers.  It is only used by Model 2 drivers.

> I tried to remove all code in drvMotorAsyn.c, but then I ran into this error when compiling/linking an IOC with ethercatmc:
```
Undefined symbols for architecture x86_64:
"_pvar_func_motorRegister", referenced from:
registrar(motorRegister)
```

If you don't build with drvMotorAsyn.c (removing support for Model 2 drivers) then you must remove this line from motorSupport.dbd:
```
registrar(motorRegister)
```
because drvMotorAsyn.c contains these lines:
```
drvMotorAsyn.c:void motorRegister(void)
drvMotorAsyn.c:epicsExportRegistrar(motorRegister);
```

kmpeters, 2020-08-12:
> There is a Model 3 version of the simulation motor, correct?

There is a model-3 version of the simulation motor.  There is no model-3 configuration in the example motorSimIOC, nor is there an example configure in the xxx module, so I haven't seen model-3 simulated motors in use yet.

prjemian, 2023-04-06:
Still a [problem](https://github.com/prjemian/epics-docker/pull/55#issuecomment-1499166784) today.  What is the recommended fix?

kmpeters, 2023-04-06:
@prjemian, I added databases to work around the problem in PR #197

kmpeters, 2023-04-06:
Eventually I'd like to see all the model-2 drivers replaced with model-3 drivers or retired.

kmpeters, 2023-04-06:
"Fixed" by #197

closed 2023-04-06 by kmpeters