epics-modules/motor #173: Latest motor (R7-2-1) does not build with latest asyn (R4-41)

MarkRivers, 2021-01-29:
The latest tag of motor, 7-2-1, from May 2020 does not build with asyn R4-41 because sharelib.h needs to be added to several files.  I fixed this in September 2020.  Could we get a new release?

kmpeters, 2021-02-01:
Yes.

kmpeters, 2021-02-01:
I'm trying and failing to reproduce the problem that was fixed by this commit using base-3.15.6:

https://github.com/epics-modules/motor/commit/173355c5ea6130609182269be2e1716341dccaf2

Does the problem only occur when motor-R7-2-1 and asyn-R-4-41 is built against base 7.0?

MarkRivers, 2021-02-01:
The problem is that epicsExport.h is not included in asynMotorController.h but it uses epicsShareClass.

EPICS base itself no longer uses epicsShareClass in 7.0, so it is likely that is was being indirectly included in 3.15.6 and no longer is.

kmpeters, 2021-02-01:
I created a bugfix branch for Pete to test before I create a 7-2-2 release:

https://github.com/epics-modules/motor/tree/R7-2-1-bugfix

The 7-3 release is going to take longer.

prjemian, 2021-02-01:
while running `make build` (from $SUPPORT), this error output on the console:

```
/usr/bin/g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE           -D_X86_64_  -DUNIX  -Dlinux      -O3   -Wall      -mtune=generic      -m64 -fPIC -I. -I../O.Common -I. -I. -I.. -I/opt/synApps/support/motor-R7-2-1-bugfix/include/compiler/gcc -I/opt/synApps/support/motor-R7-2-1-bugfix/include/os/Linux -I/opt/synApps/support/motor-R7-2-1-bugfix/include   -I/opt/synApps/support/motor-R7-2-1-bugfix/include   -I/opt/synApps/support/asyn-R4-41/include   -I/opt/synApps/support/seq-2-2-8/include   -I/opt/synApps/support/busy-R1-7-3/include   -I/opt/synApps/support/ipac-2-16/include -I/opt/base/include/compiler/gcc -I/opt/base/include/os/Linux -I/opt/base/include           -MM -MF ImsMDrivePlusMotorController.d  ../ImsMDrivePlusMotorController.cpp
make -C ./motorKohzu install
make[3]: Entering directory '/opt/synApps/support/motor-R7-2-1-bugfix/modules/motorKohzu'
make -C ./configure install
make[4]: Entering directory '/opt/synApps/support/motor-R7-2-1-bugfix/modules/motorKohzu/configure'
In file included from ../testSFTPUpload.cpp:3:
../xpsSFTPUpload.h:3:1: error: 'epicsShareFunc' does not name a type
    3 | epicsShareFunc int xpsSFTPUpload(std::string IPAddress, std::string trajectoryDirectory, std::string fileName,
      | ^~~~~~~~~~~~~~
../testSFTPUpload.cpp: In function 'int main(int, char**)':
../testSFTPUpload.cpp:11:16: error: 'xpsSFTPUpload' was not declared in this scope
   11 |   int status = xpsSFTPUpload("164.54.160.71", "/Admin/Public/Trajectories", "TrajectoryScan.trj",
      |                ^~~~~~~~~~~~~
/usr/bin/g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE           -D_X86_64_  -DUNIX  -Dlinux      -O3   -Wall      -mtune=generic      -m64 -fPIC -I. -I../O.Common -I. -I. -I.. -I/opt/synApps/support/motor-R7-2-1-bugfix/include/compiler/gcc -I/opt/synApps/support/motor-R7-2-1-bugfix/include/os/Linux -I/opt/synApps/support/motor-R7-2-1-bugfix/include   -I/opt/synApps/support/motor-R7-2-1-bugfix/include   -I/opt/synApps/support/asyn-R4-41/include   -I/opt/synApps/support/seq-2-2-8/include   -I/opt/synApps/support/busy-R1-7-3/include   -I/opt/synApps/support/ipac-2-16/include -I/opt/base/include/compiler/gcc -I/opt/base/include/os/Linux -I/opt/base/include           -MM -MF drvMDrive.d  ../drvMDrive.cc
perl -CSD /opt/base/bin/linux-x86_64/makeMakefile.pl O.linux-x86_64 ../..
mkdir -p O.Common
make -C O.linux-x86_64 -f ../Makefile TOP=../.. \
    T_A=linux-x86_64 install
make[5]: Entering directory '/opt/synApps/support/motor-R7-2-1-bugfix/modules/motorKohzu/configure/O.linux-x86_64'
```

prjemian, 2021-02-01:
The more complete output from rebuilding the docker image is here: https://github.com/prjemian/epics-docker/blob/623e8ec059ad7e913c6f788ea3915791091a51fc/v1.1/n3_synApps/build-log.txt

prjemian, 2021-02-01:
The snippet above was from the `v1.1/n3_synApps/build-log.txt` file at lines 10372-10390

kmpeters, 2021-02-01:
I think that's actually a different error.  Comment out the SUPPORT definition in the following submodules and see if the build succeeds:

./modules/motorAerotech/configure/RELEASE:#SUPPORT=$(MOTOR)/..
./modules/motorNewport/configure/RELEASE:#SUPPORT = $(MOTOR)/..
./modules/motorOms/configure/RELEASE:#SUPPORT=$(MOTOR)/..
./modules/motorParker/configure/RELEASE:#SUPPORT=$(MOTOR)/..
./modules/motorScriptMotor/configure/RELEASE:#SUPPORT=$(MOTOR)/..

The definition is need for screen conversions but it breaks the build.

prjemian, 2021-02-01:
Made those 5 edits, then `make rebuild` in `$MOTOR/`, this error:

```
/usr/bin/g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE         -Wno-write-strings   -D_X86_64_  -DUNIX  -Dlinux      -O3   -Wall      -mtune=generic      -m64  -I. -I../O.Common -I. -I. -I.. -I/opt/synApps/support/motor-R7-2-1-bugfix/include/compiler/gcc -I/opt/synApps/support/motor-R7-2-1-bugfix/include/os/Linux -I/opt/synApps/support/motor-R7-2-1-bugfix/include   -I/opt/synApps/support/motor-R7-2-1-bugfix/include   -I/opt/synApps/support/asyn-R4-41/include   -I/opt/synApps/support/seq-2-2-8/include   -I/opt/synApps/support/busy-R1-7-3/include   -I/opt/synApps/support/ipac-2-16/include -I/opt/base/include/compiler/gcc -I/opt/base/include/os/Linux -I/opt/base/include           -c ../testSFTPUpload.cpp
In file included from ../testSFTPUpload.cpp:3:
../xpsSFTPUpload.h:3:1: error: 'epicsShareFunc' does not name a type
    3 | epicsShareFunc int xpsSFTPUpload(std::string IPAddress, std::string trajectoryDirectory, std::string fileName,
      | ^~~~~~~~~~~~~~
../testSFTPUpload.cpp: In function 'int main(int, char**)':
../testSFTPUpload.cpp:11:16: error: 'xpsSFTPUpload' was not declared in this scope
   11 |   int status = xpsSFTPUpload("164.54.160.71", "/Admin/Public/Trajectories", "TrajectoryScan.trj",
      |                ^~~~~~~~~~~~~
make[5]: *** [/opt/base/configure/RULES_BUILD:249: testSFTPUpload.o] Error 1
```

Same error.

kmpeters, 2021-02-01:
I need to cherry-pick this commit too:

https://github.com/epics-motor/motorNewport/commit/070c93e1aa1a4dea39829f7dcb21489cfde582a1

I'll push updates to the bugfix branch momentarily

kmpeters, 2021-02-01:
The bugfix branch has been updated.  You'll need to do a ``git submodule update modules/motorNewport`` after updating the bugfix branch.

prjemian, 2021-02-01:
I'll do a complete rebuild from the `assemble_synApps.sh` step which will pick up the changes.  Should I make the five additional changes [above](https://github.com/epics-modules/motor/issues/173#issuecomment-771219059)?

kmpeters, 2021-02-01:
I would.

prjemian, 2021-02-01:
I went ahead before your reply.  All compiled with no errors.  Activated `motors.iocsh` (copy from examples/motors.iocsh and edit into st.cmd.Linux) and started IOC.  Separate client is able to `caget xxx:m1` with success.  Any other tests?

prjemian, 2021-02-01:
But, sadly, the `xxx:m1` PV does not respond to any `caput` commands to change position.  That is, `caput xxx:m1 1` does not produce a CA monitor update to show the new value in `xxx:m1.VAL` *and* there is no change in the `xxx:m1.RBV` value.  It remains at zero.  A `caget xxx:m1` request will show the new value but a GUI that is watching (`pvview xxx:UPTIME xxx:m1.{RBV,VAL,MOVN} &`) shows 0 values for the motor PVs.  The `xxx:UPTIME` PV is updating, as expected, at 1 Hz`.

prjemian, 2021-02-01:
Changed my build to use `master` branch and get same lack of response.  Maybe my setup is not right?

```
root@poof:/opt/synApps/support/xxx-R6-2/iocBoot/iocxxx# ls
Makefile    auto_positions.req  bootParms     envPaths      saveData.req    softioc        st.cmd.Win32    substitutions
README      auto_settings.req   common.iocsh  examples      scan1Menu.req   st.cmd.Cygwin  st.cmd.Win64
SGMenu.req  autosave            dbl-all.txt   motors.iocsh  settings.iocsh  st.cmd.Linux   st.cmd.vxWorks
root@poof:/opt/synApps/support/xxx-R6-2/iocBoot/iocxxx# git diff st.cmd.Linux 
diff --git a/iocBoot/iocxxx/st.cmd.Linux b/iocBoot/iocxxx/st.cmd.Linux
index 0c0a74d..a8a5856 100644
--- a/iocBoot/iocxxx/st.cmd.Linux
+++ b/iocBoot/iocxxx/st.cmd.Linux
@@ -13,6 +13,7 @@ iocxxxLinux_registerRecordDeviceDriver(pdbbase)
 
 < settings.iocsh
 < common.iocsh
+< motors.iocsh
 
 ###############################################################################
 iocInit
```

prjemian, 2021-02-01:
```
iocxxx> dbpr xxx:m1
ASG :               ATHM: 0             CDIR: 0             CNEN: Disable       
DESC: M0X           DHLM: 100           DIFF: 0             DISA: 0             
DISP: 0             DISV: 1             DLLM: -100          DMOV: 1             
DRBV: 0             DVAL: 0             HLM : 100           HLS : 0             
LLM : -100          LLS : 0             LVIO: 0             MOVN: 0             
NAME: xxx:m1        OFF : 0             RBV : 0             RDIF: 0             
REP : 0             RHLS: 0             RLLS: 0             RLV : 0             
RMP : 0             RRBV: 0             RVAL: 0             SEVR: NO_ALARM      
STAT: NO_ALARM      TDIR: 0             TPRO: 0             VAL : 0             
VERS: 7.2           
iocxxx> dbpr xxx:m1, 9
ACCL: 0.2           ACKS: NO_ALARM      ACKT: YES           ADEL: 0             
ALST: 0             ASG :               ASP : PTR (nil)     ATHM: 0             
BACC: 0.2           BDST: 0             
BKLNK: ELL 10 [0x561e12fc8f10 .. 0x561e12fca268]            BKPT: 00            
BVEL: 1             CARD: 0             CBAK: PTR 0x561e13257860                
CDIR: 0             CNEN: Disable       DCOF: 0             DESC: M0X           
DHLM: 100           DIFF: 0             DINP: CONSTANT      DIR : Pos           
DISA: 0             DISP: 0             DISS: NO_ALARM      DISV: 1             
DLLM: -100          DLY : 0             DMOV: 1             DOL : CONSTANT      
DPVT: PTR 0x561e13257890                DRBV: 0             
DSET: PTR 0x7f8fbc2225e0                DTYP: asynMotor     DVAL: 0             
EGU : mm            ERES: 2.5e-04       EVNT:               FLNK: CONSTANT      
FOF : 0             FOFF: Variable      FRAC: 1             HHSV: NO_ALARM      
HIGH: 0             HIHI: 0             HLM : 100           HLS : 0             
HLSV: NO_ALARM      HOMF: 0             HOMR: 0             HOPR: 0             
HSV : NO_ALARM      HVEL: 0.1           ICOF: 0             IGSET: 0            
INIT:               JAR : 5             JOGF: 0             JOGR: 0             
JVEL: 1             LCNT: 0             LDVL: 0             LLM : -100          
LLS : 0             LLSV: NO_ALARM      LOCK: NO            LOLO: 0             
LOPR: 0             LOW : 0             LRLV: 0             LRVL: 0             
LSET: PTR 0x561e13026db0                LSPG: Go            LSV : NO_ALARM      
LVAL: 0             LVIO: 0             MDEL: 0             MIP : 0             
MISS: 0             MLIS: ELL 4 [0x7f8f60102ab8 .. 0x7f8f6454b980]              
MLOK: e0 49 06 13 1e 56 00 00           MLST: 0             MMAP: 0             
MOVN: 0             MRES: 2.5e-04       MSTA: 0             NAME: xxx:m1        
NMAP: 0             NSEV: NO_ALARM      NSTA: NO_ALARM      NTM : YES           
NTMF: 2             OFF : 0             OMSL: supervisory   OUT : INST_IO @     
PACT: 1             PCOF: 0             PHAS: 0             PINI: NO            
POST:               PP  : 0             PPN : PTR (nil)     PPNR: PTR (nil)     
PREC: 5             PREM:               PRIO: LOW           PROC: 0             
PUTF: 0             RBV : 0             RCNT: 0             RDBD: 2.5e-04       
RDBL: CONSTANT      RDES: PTR 0x561e12cc1fc0                RDIF: 0             
REP : 0             RHLS: 0             RINP: CONSTANT      RLLS: 0             
RLNK: CONSTANT      RLV : 0             RMOD: Default       RMP : 0             
RPRO: 0             RRBV: 0             RRES: 0             
RSET: PTR 0x7f8fbc222400                RTRY: 10            RVAL: 0             
RVEL: 0             S   : 20            SBAK: 20            SBAS: 2             
SCAN: Passive       SDIS: DB_LINK xxx:m1_able.VAL NPP NMS   SET : Use           
SEVR: NO_ALARM      SMAX: 0             SPDB: 0             SPMG: Go            
SPVT: PTR (nil)     SREV: 200           SSET: 0             STAT: NO_ALARM      
STOO: CONSTANT      STOP: 0             STUP: OFF           SUSE: 0             
SYNC: 0             TDIR: 0             TIME: <undefined>   TPRO: 0             
TSE : 0             TSEL: CONSTANT      TWF : 0             TWR : 0             
TWV : 1             UDF : 0             UDFS: INVALID       UEIP: No            
UREV: 0.05          URIP: No            VAL : 0             VBAS: 0.1           
VELO: 1             VERS: 7.2           VMAX: 0             VOF : 0             
iocxxx> 
```

kmpeters, 2021-02-02:
motor.iocsh in xxx loads motor records but doesn't configure any motor controllers:

https://github.com/epics-modules/xxx/blob/master/iocBoot/iocxxx/examples/motors.iocsh

I'm going to declare victory and release R7-2-2

tboegi, 2021-02-02:
Not sure, if the updated OMS driver had been added in modules/ ?

But, just to be sure:
https://github.com/epics-modules/motor/pull/174

kmpeters, 2021-02-02:
@tboegi I will include that too.

kmpeters, 2021-02-02:
https://github.com/epics-modules/motor/releases/tag/R7-2-2

closed 2021-02-02 by kmpeters