epics-modules/motor #185: Include motorSymetrie in motor

kmpeters, 2022-05-09:
A place to discuss what needs to be done to include motorSymetrie in motor

kmpeters, 2022-05-09:
Do you propose waiting until the motor-record support is done to include motorSymetrie in motor?  I'm not opposed to including it before then.

It worries me a little bit that pmac depends on Diamond-specific forks of synApps modules.  I'm able to build pmac & motorSymetrie, but I don't know if the support depends on changes in Diamond's version of motor that aren't included in this module.

MarkRivers, 2022-05-24:
@kmpeters it is actually not possible to include motorSymetrie in motor.  The problem is that pmac is not part of motor, but pmac depends on motor.  If motorSymetrie is part of motor there is a circular dependency, because the pmac depends on motor and motor depends on pmac.

I have moved motorsSymetrie to be a top-level module in support, and that works fine.

There is also an issue that pmac does not work on Windows shareable, e.g. windows-x64 architecture, because it lacks the epicsShareClass definitions and export.h is in the wrong location.  I am working on fixing that, but for now I have change the motorSymetrie Makefile to only build on Linux.

We should probably close this issue.

kmpeters, 2022-05-25:
Sounds good.

closed 2022-05-25 by kmpeters