epics-modules/motor #189: Iocsh referenced by xxx module is missing

djvine, 2022-08-22:
This is the same issue as https://github.com/epics-modules/xxx/issues/26

@prjemian merged a pull request at the time but that appears to be broken. There's no iocsh directory in motor or motorSim.

The xxx module references the following to instantiate some simulated motors:
iocshLoad("$(MOTOR)/iocsh/motorSim.iocsh", "INSTANCE=motorSim, HOME_POS=0, NUM_AXES=16, HIGH_LIM=32000, LOW_LIM=-32000, SUB=substitutions/motorSim.substitutions")

prjemian, 2022-08-22:
The [`docs/RELEASE.md`](https://github.com/epics-modules/motor/blob/master/docs/RELEASE.md#driver-submodules-and-noteworthy-changes-2) file states that amongst the changes for R7.2: `iocsh files are now installed at build time`.

prjemian, 2022-08-22:
The make rule source (new in R6.2.1): https://github.com/EPICS-synApps/support/blob/R6-2-1/configure/CONFIG_IOCSH

kmpeters, 2022-08-25:
motorSim.iocsh can also appear to be missing if motorMotorSim is built outside of the motor module.  In this case it would be found here:
```
$(MOTOR_MOTORSIM)/iocsh/motorSim.iocsh
```
Instead of inside motor:
```
$(MOTOR)/iocsh/motorSim.iocsh
```

MarkRivers, 2022-08-25:
Note that motorSim.iocsh is using the obsolete and deprecated Model 2 driver.  It should be updated to use the Model 3 driver.  I have just created a PR to update motorMotorSim/iocs/motorSimIOC/iocBoot/iocMotorSim to use the Model 3 driver, but I did not update motorSim.iocsh.

kmpeters, 2023-05-15:
I added motorSim8x.iocsh to motorMotorSim recently (https://github.com/epics-motor/motorMotorSim/commit/32547ab47953e08ef06ed291cd699444af1e7987), which uses the model-3 support and can be sourced like the older motorSim.iocsh.  It is installed to the motor's iocsh dir when motorMotorSim is built as a submodule.  It is installed to motorMotorSim's iocsh dir when motorMotorSim is built outside of motor.

closed 2023-05-15 by kmpeters