epics-modules/motor #194: Github Actions builds fail on Windows with base 7.0 development branch

kmpeters, 2022-11-30:
Switch to a release of base instead.

kmpeters, 2022-11-30:
It looks like the github actions builds were always using base 7.0.  I still don't know why.

prjemian, 2022-11-30:
https://github.com/epics-modules/motor/blob/302d4d61fc374dfd23c8a7dcdadeb4937cce40bc/.github/workflows/ci-scripts-build-full.yml#L44 https://github.com/epics-modules/motor/blob/302d4d61fc374dfd23c8a7dcdadeb4937cce40bc/.github/workflows/ci-scripts-build-full.yml#L60 ...

Looks like this is a manually-controlled item.

kmpeters, 2022-11-30:
Yes, but the windows config looks like this:

https://github.com/epics-modules/motor/blob/302d4d61fc374dfd23c8a7dcdadeb4937cce40bc/.github/workflows/ci-scripts-build.yml#L62-L67

kmpeters, 2022-11-30:
There still might be something wrong with my github actions config.

prjemian, 2022-11-30:
In the [logs for the Win build](https://pipelines.actions.githubusercontent.com/serviceHosts/d6d51220-a37a-4a5e-b217-8bb6ad75e815/_apis/pipelines/1/runs/16/signedlogcontent/3?urlExpires=2022-11-30T17%3A03%3A11.3698939Z&urlSigningMethod=HMACV1&urlSignature=6vJ5fIInXZoikttD41gBiKv5wvauw%2FKWiR9FPJ3c%2Bw0%3D), here's where base-7.0 was identified:

```
2022-11-30T15:27:29.3791751Z D:\a\motor\motor>call "C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Auxiliary\Build\vcvarsall.bat" amd64 
2022-11-30T15:27:30.0013274Z **********************************************************************
2022-11-30T15:27:30.0013647Z ** Visual Studio 2019 Developer Command Prompt v16.11.21
2022-11-30T15:27:30.0014566Z ** Copyright (c) 2021 Microsoft Corporation
2022-11-30T15:27:30.0015114Z **********************************************************************
2022-11-30T15:27:31.9176878Z [vcvarsall.bat] Environment initialized for: 'x64'
2022-11-30T15:27:32.0426086Z [36;1mBuild using vs2019 compiler on windows (x64) hosted by github-actions[0m
2022-11-30T15:27:32.0426793Z [36;1mPython setup[0m
2022-11-30T15:27:32.0427294Z 3.7.9 (tags/v3.7.9:13c94747c7, Aug 17 2020, 18:58:18) [MSC v.1900 64 bit (AMD64)]
2022-11-30T15:27:32.0427748Z PYTHONPATH
2022-11-30T15:27:32.0428108Z   D:\a\motor\motor\.ci
2022-11-30T15:27:32.0428572Z   D:\a\_actions\mdavidsaver\ci-core-dumper\master
2022-11-30T15:27:32.0429011Z   D:\a\motor\motor
2022-11-30T15:27:32.0429473Z   C:\hostedtoolcache\windows\Python\3.7.9\x64\python37.zip
2022-11-30T15:27:32.0429963Z   C:\hostedtoolcache\windows\Python\3.7.9\x64\DLLs
2022-11-30T15:27:32.0430449Z   C:\hostedtoolcache\windows\Python\3.7.9\x64\lib
2022-11-30T15:27:32.0430942Z   C:\hostedtoolcache\windows\Python\3.7.9\x64
2022-11-30T15:27:32.0431439Z   C:\hostedtoolcache\windows\Python\3.7.9\x64\lib\site-packages
2022-11-30T15:27:32.0431901Z platform = win-amd64
2022-11-30T15:27:32.0432333Z [36;1mAvailable Visual Studio versions[0m
2022-11-30T15:27:32.0432909Z vs2019 in C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Auxiliary\Build\vcvarsall.bat
2022-11-30T15:27:32.0433491Z vs2015 in C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\vcvarsall.bat
2022-11-30T15:27:32.0434593Z ##[group][33;1mLoading setup files[0m
2022-11-30T15:27:32.0435049Z Opening setup file .ci-local\modules.set
2022-11-30T15:27:32.0435497Z Opening setup file .ci\defaults.set
2022-11-30T15:27:32.0436074Z ##[endgroup]
2022-11-30T15:27:32.2047781Z ##[group][33;1mChecking/cloning dependencies[0m
2022-11-30T15:27:32.6985176Z 8969a952e4853dfe45156eab2a7176498fa5e86b	refs/heads/7.0
2022-11-30T15:27:32.7045992Z Cloning 7.0 of dependency BASE into C:\Users\runneradmin\.cache\base-7.0
2022-11-30T15:27:57.9109209Z commit 8969a952e4853dfe45156eab2a7176498fa5e86b
```

kmpeters, 2022-11-30:
Does that make it obvious to you what is wrong with the config?  I'm not seeing the mistake.

prjemian, 2022-11-30:
Not so obvious to me, which is why I support Andrew's suggestion.

kmpeters, 2022-11-30:
I'll try that.

anjohnson, 2022-11-30:
This error from your build is what I need to fix in Base, so using 7.0.7 should fix the problem:
```
dbdExpand.pl: No input files for dlload.dbd
```

kmpeters, 2022-11-30:
This change fixed the problem of the specified versions of base being ignored:

https://github.com/epics-modules/motor/pull/195/commits/566f47073682973f185ee76ee7ddfbeab74bb442

I'm not including the base version in the modules.set file.  I suspect I based the ci-scripts-build.yml file on ci-scripts-build.yml.example-full from the ci-scripts module, which also doesn't set the BASE env var based on matrix.base:
```bash
[host ~/development/epics-modules/ci-scripts/github-actions]$ ls -l
total 16
-rw-rw-r-- 1 user group 6207 Nov 30 12:54 ci-scripts-build.yml.example-full
-rw-rw-r-- 1 user group 1956 Nov 30 12:54 ci-scripts-build.yml.example-mini
-rw-rw-r-- 1 user group 2605 Nov 30 12:54 README.md
[host ~/development/epics-modules/ci-scripts/github-actions]$ grep -ir matrix.base .
./ci-scripts-build.yml.example-mini:      BASE: ${{ matrix.base }}
[host ~/development/epics-modules/ci-scripts/github-actions]$ 
```

commit 34474ed9 (2022-11-30): Merge pull request #195 from epics-modules/issue194

Fix for Github Actions builds using the wrong version of base: set the BASE env var using matrix.base.

Fixes #194
closed 2022-11-30 by kmpeters