epics-modules/motor #205: RHLM, RLLM and MRES < 0 doesn't work

tboegi, 2023-05-22:
(Currently testing on commit d80b006a39de2853090a1d9574bb0297c2b92e0c)

The introduction of RHLM and RLLM did not consider the use case
when MRES < 0:
When commanded "forwards", the motor moves "backwards".
The soft limits needs to be swapped, depending on the change,
either RHLM vs RLLM or DHLM vs DLLM.

As a rule of thumb, RHLM should be updated when SET_HIGH_LIMIT
is send to the driver:

`        if (command == SET_HIGH_LIMIT)
        {
            pmr->rhlm = tmp_raw;
`

I am preparing a patch @kmpeters, should be ready the next days

tboegi, 2023-05-22:
https://github.com/epics-modules/motor/pull/206

rerpha, 2023-05-24:
Sorry @tboegi , was an oversight on my part! Thanks for fixing