epics-modules/motor #219: Increase arbitrary MAX_AXIS limit

kmpeters, 2024-07-11:
`MAX_AXIS` was last increased (from 10 to 32) twenty years ago:

https://github.com/epics-modules/motor/commit/141daff8174a2dbac1ab7162b8c5ca258fcd0c3a

Controllers that support 64 axes are increasingly common.  `MAX_AXIS` should probably be increase to 128.

`MAX_AXIS` only appears to be used in one location:

https://github.com/epics-modules/motor/blob/81662468d0ab2ff0896eb6fbfab214165122ceb2/motorApp/MotorSrc/motordrvCom.h#L178

MarkRivers, 2024-07-12:
And the one place it is used is only for Model 1 drivers which should never be used for new drivers.

kmpeters, 2024-07-12:
@MarkRivers, are you aware of any 32-axis limitations for model-3 drivers?  I didn't find any in the asyn motor support or my grepping of the asyn module.

Axilion has reported a 32-axis limitation with motorAcsMotion, but the driver-imposed maximun number of axes has been 64 for the last four years.

MarkRivers, 2024-07-13:
I don’t know of any, but I have searched.