epics-modules/motor #222: No license?

ralphlange, 2024-12-13:
This repo does not contain a license file, nor does the documentation give good hints.

keenanlang, 2025-05-21:
Missed a 2 in the commit message, but addressed in https://github.com/epics-modules/motor/commit/26efd2f044f6281b5c5889d604cd43f84b4bf615

kmpeters, 2026-02-10:
@anjohnson, is linking to the synApps license sufficient?

https://www.aps.anl.gov/BCDA/synApps-license

If not, how do I determine who the copyright holders of the motor module are for the [EPICS open license](https://epics-controls.org/epics-open-license)?  I know that UChicago Argonne LLC. should be listed as a copyright holder, but the motor module existed many years before I joined the lab.

Are there guidlines on how frequently the copyright years in the license should be updated?

https://github.com/epics-base/epics-base/blob/4fe5f7538722950fbdd61aad0a87eaf9896a9656/LICENSE#L1-L6

anjohnson, 2026-02-10:
If you only provide a link without including a copy of the license terms and that link target gets moved, changed or removed in the future, it becomes harder for someone with only a copy of your source code containing the link to know what license terms apply to the code.

EPICS Base provides a `LICENSE` file, and we now include an SPDX License Identifier line in a header comment in all our source files:
```C
/*************************************************************************\
* Copyright (c) ...
* SPDX-License-Identifier: EPICS
* EPICS BASE is distributed subject to a Software License Agreement found
* in file LICENSE that is included with this distribution.
\*************************************************************************/
```

The EPICS Open License is [registered with SPDX](https://spdx.org/licenses/EPICS.html), you should use the same ID (they don't want duplicates when the text is identical). See [here](https://spdx.github.io/using/license-id-in-source/) for more info about how to embed SPDX license IDs.

The copyrights listed in a `LICENSE` file normally refer to who owns the license text itself and when it was written, which are normally totally different to the software it covers. For example every copy of the GNU GPL `COPYING` file includes a Free Software Foundation copyright notice at the top with terms that say you can only distribute it unchanged. When EPICS was open-sourced some of the other organizations didn't understand that and I didn't push back against their lawyers. We include the copyright notice in each source file as shown above to allow contributors to each file to add their owner and date in their PR, although almost nobody ever actually does that.

kmpeters, 2026-02-10:
Was a command line tool like [reuse](https://reuse.software/) used to add the license header to all the source files in base?

kmpeters, 2026-02-10:
I created a branch to add the license.  I've only added the top-level LICENSE file so far:

https://github.com/epics-modules/motor/blob/38d3779c5f0889624456ca79db2a81cb628f9bca/LICENSE

anjohnson, 2026-02-11:
I'd been encouraging people to add the standard header to our source files for many years (Janet Anderson did a lot of them, and she retired quite a few years ago).

Those reuse tools look interesting, but I don't believe they were available back in June 2020 when I did most of the work to add the license ID (we don't tag Makefiles though). The `SPDX-FileCopyrightText` tag is a newer addition which we don't have in Base; they seem to be suggesting that license texts should live in a top-level `LICENSES` directory, which might be worth doing if you want to be compliant with that part of their spec.

tboegi, 2026-02-12:
Does  4c means that every fork of motor, which had mad changes, should state that.
Where should that information go ?
It is already visible in `git log`, isnt it ?

anjohnson, 2026-02-12:
I also believe that the contents of a CI config file are almost completely functional, and those aspects of a file are not eligible for copyright anyway. That's why we didn't add notices to the Makefiles in EPICS.

kmpeters, 2026-02-12:
If your project contains a license that isn't registered, it is detected:

```bash
❯ ls -l LICENSES
total 8
-rw-rw-r-- 1 kpetersn aesbc 3487 Feb 11 11:40 EPICS.txt
-rw-rw-r-- 1 kpetersn aesbc 3487 Feb 11 11:40 NOT_EPICS.txt
❯ reuse lint
reuse.project - WARNING - Could not resolve SPDX License Identifier of LICENSES/NOT_EPICS.txt, resolving to NOT_EPICS. Make sure the license is in the license list found at <https://spdx.org/licenses/> or that it starts with 'LicenseRef-', and that it has a file extension.
# BAD LICENSES

'NOT_EPICS' found in:
* LICENSES/NOT_EPICS.txt

# UNUSED LICENSES

The following licenses are not used:
* NOT_EPICS

# SUMMARY

* Bad licenses: NOT_EPICS
* Deprecated licenses: 0
* Licenses without file extension: 0
* Missing licenses: 0
* Unused licenses: NOT_EPICS
* Used licenses: EPICS
* Read errors: 0
* Files with copyright information: 70 / 70
* Files with license information: 70 / 70

Unfortunately, your project is not compliant with version 3.3 of the REUSE Specification :-(

# RECOMMENDATIONS

* Fix bad licenses: At least one license in the LICENSES directory and/or
  provided by 'SPDX-License-Identifier' tags is invalid. They are either not
  valid SPDX License Identifiers or do not start with 'LicenseRef-'. FAQ about
  custom licenses: https://reuse.software/faq/#custom-license
* Fix unused licenses: At least one of the license text files in 'LICENSES' is
  not referenced by any file, e.g. by an 'SPDX-License-Identifier' tag. Please
  make sure that you either tag the accordingly licensed files properly, or
  delete the unused license text if you are sure that no file or code snippet is
  licensed as such.
```